keyboardio/Kaleidoscope #263: Strange consumer ctrl key definitions

noseglasses, 2017-12-06:
In the following line 
https://github.com/keyboardio/Kaleidoscope/blob/c060c5ef7e76f5bce5524d66921a1179724b260d/src/key_defs_consumerctl.h#L131
when the macro `Consumer_FanSpeed` initializes a `Key` object, a `.keyCode` value is supplied that is based on `HID_CONSUMER_FAN_SPEED` which is actualy `x101` and thus greater than what a `uint8_t` value can store.

The corresponding value is redundantly defined here:
https://github.com/keyboardio/Kaleidoscope/blob/c060c5ef7e76f5bce5524d66921a1179724b260d/src/HIDTables.h#L423
and here:
https://github.com/keyboardio/KeyboardioHID/blob/df02dc93865ab85c1df73eb3b7d133c21f70c4b5/src/HIDTables.h#L424

When the macro is invoked, gcc reports  a narrowing conversion from `int` to `uint8_t` when build for x86. Probably the respective macro was never used, so the error did not manifest.

algernon, 2017-12-06:
It does *not* initialise `.keyCode`, not directly. The [CONSUMER_KEY macro](https://github.com/keyboardio/Kaleidoscope/blob/8c250fe028d934a0d90e8454f94189915e4fa6e6/src/key_defs.h#L132) merges its arguments into `.raw`, which is an `uint16_t`, and can safely hold the value.

noseglasses, 2017-12-06:
Sorry, my bad. I worked with a slightly outdated version, the following commit fixed the problem:
https://github.com/keyboardio/Kaleidoscope/commit/8ab23033e7a7c13d772dd2e68bb6e6645fac3e8c#diff-19ffed0b6e395a10f285d61e83c85f8f

This issue can be closed.

algernon, 2017-12-06:
\o/

Glad it is something already fixed =)

closed 2017-12-06 by algernon