keyboardio/Kaleidoscope #265: Keyboard repeats keypresses forever

SwartzCr, 2017-12-11:
On my machine alt+space is a command used to open a scratch window.
In certain cases when I type this on my keyboardio keyboard it will continue to press that key combination forever, causing the window to blink in and out of existence until space or alt is pressed again, at which point it will stop.
This happens pretty regularly and seems likely to be a software issue.
I sent videos into Jesse, but don't want to post them on github since they contain personal information displayed on my screen at the time

imrekoszo, 2018-01-16:
Are you using the right half of the keyboard to press alt+space? If yes, this could be related to #239

obra, 2018-01-16:
@SwartzCr - Was this with your first keyboard, your second keyboard or both?

gedankenexperimenter, 2019-05-06:
It's been a whole year since there has been any news about this. I think it would be fair to close it at this point.

closed 2019-05-10 by algernon