keyboardio/Kaleidoscope #268: Question: Why does the class Layer_ need an instance?

gedankenexperimenter, 2017-12-15:
In `layers.h`, instead of naming the class `Layer`, it's named `Layer_`, and an instance is declared in `layers.cpp` (named `Layer`). Why is this necessary? Everything in the class is declared static, so why isn't it possible to just define it as `class Layer` and not bother with an instance?

Actually changing it would involve changing lots of cases of `Layer.*` to `Layer::*`, even in some plugins; I'm just wondering if there's a reason for it being done with a pseudo-singleton instead of just the class itself.

algernon, 2017-12-15:
It used to be a class that had a non-static member (`begin`, and it was a descendant of `KaleidoscopePlugin`). It was left this way since then.

The one advantage I see with the instance is that its usage is consistent with pretty much everything else: you don't have to write `Layer::foo`, while you write `Kaleidoscope.bar` or `OneShot.something`. This is, in my opinion, a strong reason to keep the (mostly useless) instance.

gedankenexperimenter, 2017-12-15:
As a followup question, why does `class Kaleidoscope` need an instance?

closed 2018-01-25 by gedankenexperimenter