keyboardio/Kaleidoscope #269: ModemManager warning prints twice regardless of service being disabled

leavesofgrass, 2017-12-17:
When flashing my firmware, I get the warning:

`"Yikes. ModemManager wants to pwn your keyboard`

printed a few times before the programmer connects.

Adding 
`use Data::Dumper; print scalar Dumper $devices{$path};` to the perl script rendered this output:

`Yikes. ModemManager wants to pwn your keyboard at /home/jon/kaleidoscope/Arduino-Boards/libraries/Kaleidoscope/bin/find-device-port-linux-udev line 25.
/home/jon/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder: 243: [: $VAR1: unexpected operator
stty: '$VAR1': No such file or directory
/home/jon/Arduino/hardware/keyboardio/avr/build-tools/makefiles//rules.mk:75: recipe for target 'flash' failed
make: *** [flash] Error 1`

algernon, 2018-10-20:
I think we're only printing once now. We do print the warning even when `ModemManager` the service is disabled, because we don't try to detect if its running: we try to detect whether the udev rules for the keyboard would stop ModemManager to jump on the serial port, if it were running. I think this is an acceptable compromise: having the udev rules present even when ModemManager is not used doesn't hurt.

I think we could resolve this...

obra, 2020-06-08:
Resolving this, as I believe the issue is resolved. Please feel free to reopen if it's not

closed 2020-06-08 by obra