keyboardio/Kaleidoscope #273: Key_NonUSPound doesn't print anything

mattvenn, 2017-12-20:
I want to print the # and ~ characters. xev shows nothing when the key is pressed.

gedankenexperimenter, 2017-12-20:
Just in case: that should be `Key_NonUsPound` – the `s` is not a capital.

gedankenexperimenter, 2017-12-20:
I can confirm that `Key_NonUsPound` is also non-functional on macOS.

gedankenexperimenter, 2017-12-20:
Could this be an OS keymap issue? The key in question is between `Z` & `shift` on some keyboards, isn't it?

algernon, 2017-12-20:
@mattvenn said it works on another keyboard of his. I'm now verifying if it is in the report, then will check OS keymap. That was my first suspicion too, but if it works on another board...?

algernon, 2017-12-20:
I can verify that the key correctly ends up in the report, so this is likely an OS-side issue.

algernon, 2017-12-20:
Ooon the other hand, the same code on the ErgoDox EZ works, with the same OS keymap. So perhaps there IS something on the firmware side that we can do.

I suspect HID Descriptors.

algernon, 2017-12-20:
Mind you, using an US QWERTY OS layout, it comes across as `\`. Using US International, it becomes `#`. However, `\` becomes `#` on US International too.

obra, 2017-12-20:
https://github.com/keyboardio/KeyboardioHID/blob/master/src/MultiReport/Keyboard.cpp#L59

This is something we inherited from the driver we learned our
cross-platform NKRO tricks from.

algernon, 2017-12-20:
```
  // Padding 3 bits
  // To skip HID_KEYBOARD_NON_US_POUND_AND_TILDE, which causes
  // Linux to choke on our driver.
```

From [KeyboardioHID](https://github.com/keyboardio/KeyboardioHID/blob/991f8f28ce5846b5bac66c15135baf663e636b00/src/MultiReport/Keyboard.cpp#L59-L64).

I'll have a go at removing this, and merging the usages. That should fix this issue, and make our descriptor a few bytes smaller too.

obra, 2017-12-20:
> 
>   // Padding 3 bits
>   // To skip HID_KEYBOARD_NON_US_POUND_AND_TILDE, which causes
>   // Linux to choke on our driver.
> From KeyboardioHID.
> 
> I'll have a go at removing this, and merging the usages. That should fix this issue, and make our descriptor a few bytes smaller too.

Er. It will need extensive testing on a range of linuxes and androids. What @triplehaata told me was that without that skip, some linux kernels don’t cope with our driver.

algernon, 2017-12-20:
I tested Linux 2.6.32, 3.something, 4.9, and they all work the same way: they detect the keyboard, everything works. `NON_US_POUND` will be translated to `BACKSLASH` though, by the kernel (as far as I see - skipping the kernel drivers shows them as separate bits in the report, but `evemu-record` shows both as the same code, and `evemu-describe` only shows `BACKSLASH`).

So I couldn't make the kernel choke on it, nor was I able to trigger double presses or anything like it.

These issues would only occur if the keys in question were on the keymap. The Model01 factory firmware has neither. So I think not skipping `NONUS_POUND` and `CLEAR` are safe.

But to be sure, I'll look into the linux kernel too.

mattvenn, 2017-12-21:
I'm on Ubuntu 16

algernon, 2017-12-21:
keyboardio/KeyboardioHID#23 should have a "fix" then. Mind you, Linux will treat `NonUS_Pound` as `Backslash`, so you won't have an additional `#/~` key :( You'll just have two backslashes.

This is something deeply rooted in the kernel, something that has been there for at least 13 years. Changing it to treat the two separately would require change to the kernel, and everything above it to support it. The chances of that happening in my lifetime are slim to none, I believe.

mattvenn, 2018-01-24:
My \ key now is printing #. Maybe that happened as a result of https://github.com/keyboardio/KeyboardioHID/pull/23. So then I used a key with no assigned keycode and used xmodmap to map it to \. So now I have both!

algernon, 2018-01-30:
I think we can safely close this now.

closed 2018-01-30 by algernon