keyboardio/Kaleidoscope #277: Port detection - /dev/serial doesn't exist

nivekuil, 2017-12-27:
Running Arch Linux. FWIW the serial port detection logic has never worked on my desktop (a Z170 chipset), but did work on my Haswell laptop.

obra, 2017-12-27:
Kevin,

Does your desktop...not use udev?

nivekuil, 2017-12-28:
It does - it's default on Arch and I wouldn't have a reason to change that. I don't think I have any udev rules that might break this. For context:
```
[~/Arduino/Model01-Firmware] $ ./make.sh
...
Press ENTER when ready...
ls: cannot access '/dev/serial/by-id/': No such file or directory
Couldn't autodetect the keyboard's serial port.
If you see this message and your keyboard is connected to your computer,
it probably means that our serial port detection logic is buggy or incomplete.
Please report this issue at https://github.com/keyboardio/Kaleidoscope

```
where make.sh is the command mentioned in https://community.keyboard.io/t/problems-getting-started-reflashing-model-01-on-arch-linux/488. The repo and submodules are up to date.

obra, 2018-01-03:
> t/problems-getting-started-reflashing-model-01-on-arch-linux/488. The

Kevin,

Do you _have_ a /dev//serial? directory? Could you show us what's inside
with your keyboard connected?

Thanks!
jesse

nivekuil, 2018-01-03:
Nope. Here's what /dev/* looks like (typing this on the model 01):
```
autofs           hidraw0    kvm                 pts              snd     tty19  tty33  tty48  tty62    vcs5
block            hidraw1    log                 random           stderr  tty2   tty34  tty49  tty63    vcs6
bsg              hidraw2    mei0                rfkill           stdin   tty20  tty35  tty5   tty7     vcsa
bus              hidraw3    mem                 roccatkoneplus0  stdout  tty21  tty36  tty50  tty8     vcsa1
char             hidraw4    memory_bandwidth    rtc              tty     tty22  tty37  tty51  tty9     vcsa2
console          hpet       mqueue              rtc0             tty0    tty23  tty38  tty52  ttyS0    vcsa3
core             hugepages  net                 sda              tty1    tty24  tty39  tty53  ttyS1    vcsa4
cpu              i2c-0      network_latency     sda1             tty10   tty25  tty4   tty54  ttyS2    vcsa5
cpu_dma_latency  i2c-1      network_throughput  sda2             tty11   tty26  tty40  tty55  ttyS3    vcsa6
disk             i2c-2      null                sdb              tty12   tty27  tty41  tty56  urandom  vfio
dri              i2c-3      port                sdb1             tty13   tty28  tty42  tty57  usb      vga_arbiter
drm_dp_aux0      i2c-4      pps0                sdc              tty14   tty29  tty43  tty58  vcs      vhost-net
fb0              initctl    psaux               sdc1             tty15   tty3   tty44  tty59  vcs1     watchdog
fd               input      ptmx                sdc2             tty16   tty30  tty45  tty6   vcs2     watchdog0
full             kfd        ptp0                shm              tty17   tty31  tty46  tty60  vcs3     zero
fuse             kmsg       ptp1                snapshot         tty18   tty32  tty47  tty61  vcs4
```
And from dmesg (let me know if you want a full log):
```
[    1.705499] input: Keyboardio Model 01 as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.2/0003:1209:2301.0002/input/input1
[    1.705596] hid-generic 0003:1209:2301.0002: input,hidraw1: USB HID v1.01 Mouse [Keyboardio Model 01] on usb-0000:00:14.0-6/input2
[    1.707963] input: Keyboardio Model 01 as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.3/0003:1209:2301.0003/input/input2
[    1.770160] hid-generic 0003:1209:2301.0003: input,hidraw2: USB HID v1.01 Mouse [Keyboardio Model 01] on usb-0000:00:14.0-6/input3
[    1.940045] usb 1-14: new full-speed USB device number 4 using xhci_hcd
```

nivekuil, 2018-01-03:
If it helps, `/dev/input/by-id/usb-Keyboardio_Model_01_Ckbio01-if03-event-mouse` is the file where the keypresses are read.

RageZBla, 2018-05-28:
Same here the funny part is I could flash without any issue on my arch at work last Friday, I can't on my arch at home but I used to be able to do it so definitely something changed. :bowing_man:

RageZBla, 2018-05-28:
ignore this [comment](https://github.com/keyboardio/Kaleidoscope/issues/277#issuecomment-392399837) rebooted and `/dev/serial` was there, not sure what happened :man_shrugging:

algernon, 2018-10-20:
We have improved the serial port detection quite a bit since last December. Is this still an issue @nivekuil?

uipoet, 2018-12-20:
Unable to flash with my Google Pixel Slate's Linux container (Debian Stretch):

```zsh
make flash
BOARD_HARDWARE_PATH="/home/jamie/Arduino/hardware" /home/jamie/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash
Building output/Model01-Firmware/Model01-Firmware (0.0.0-gv1.22-43-ge56f) ...
- Size: firmware/Model01-Firmware/Model01-Firmware-0.0.0-gv1.22-43-ge56f.elf
  - Program:   24770 bytes (86.4% Full)
  - Data:       1407 bytes (55.0% Full)

To update your keyboard's firmware, hold down the 'Prog' key on your keyboard,
and then press 'Enter'.

When the 'Prog' key glows red, you can release it.

ls: cannot access '/dev/serial/by-id/': No such file or directory
Looking for USB device with vid=1209 and pid=2301
ERROR: I couldn't find a USB device matching the keyboard's USB Vendor and Device IDs

I couldn't autodetect the keyboard's serial port.

If you see this message and your keyboard is connected to your computer,
it may mean that our serial port detection logic is buggy or incomplete.
In that case, please report this issue at:
         https://github.com/keyboardio/Kaleidoscope
/home/jamie/Arduino/hardware/keyboardio/avr/build-tools/makefiles//rules.mk:80: recipefor target 'flash' failed
make: *** [flash] Error 1
```

obra, 2018-12-20:
ChromeOS doesn’t yet expose serial devices to linux containers. Until they fix that, there’s nothing we can do :/

uipoet, 2018-12-20:
Apologies, I didn't see that USB access inside the Linux containers hasn't landed in the stable channel yet.

obra, 2018-12-20:
No worries. I’m waiting for it too. Otherwise, how do you like the slate?

uipoet, 2018-12-20:
The only issue I have with the Model 01 on the Slate is when plugging in directly via USB-C to USB-C, there is an audible whistle that increases in volume proportionate to the brightness of the LEDs. With LEDs off it is the loudest and with some of the brightest modes it is the quietest. This issue is not present when I connect USB-C to USB-A on a moshi dock, presumably because it has its own power source. 

I eagerly await gpu acceleration and now USB support with Linux apps. Visual Studio Code is rather slow on my large external monitor, but pretty good on the Slate's native screen. The pen is terrible. I have to press hard sometimes to have it register taps and dragging is almost impossible. Wonder if I have a dud or it's a software bug that will right itself eventually?

I'm tempted to put it in "dev mode" to give Crouton a spin. Full Linux in this form factor might ultimately be what I'm after. Alas, desktops for Linux are such a mess. My desktop runs Gnome over NixOS so I'd probably head in that direction for consistency.

What I do not miss at all is my MacBook Pro! Gone is the heavy weight, proprietary software oddities and a useless keyboard now that I can travel with my Model 01. 👍

obra, 2020-06-08:
Closing, as I believe the original issue is long resolved.

closed 2020-06-08 by obra