keyboardio/Kaleidoscope #283: Extra characters input when using modifier flags

gedankenexperimenter, 2018-01-23:
An example, using the default Model01-Firmware:

1. Press key `fn`
2. Press key `C`
3. Press and release key `U` (`{` on `FUNCTION` layer)
4. Repeat step 3 several times
5. Release all keys

…the output I see in a terminal on both macOS and Linux is as follows:
```text
ccccccccC{C{C{C{
```

If, instead, I do this:

1. Press key `fn`
2. Press key `C`
3. Press key `shift`
4. Press and release key `O` (`[` on `FUNCTION` layer)
5. Repeat steps 3 & 4 several times
6. Release all keys

…the output is the expected:
```text
cccccccc{{{{{
```

After holding a printable key, when the next key is typed, the previous one is masked at some level by the host, so the `c` stops repeating, but when a key with the `SHIFT_HELD` flag is pressed, an extra modified character for the held key appears in the output.

gedankenexperimenter, 2018-01-23:
I'm not entirely certain because I'm having trouble getting my limited debugging tools to work the way I expect them to, but I believe this is caused by the workaround for the ChromeOS bug in [hid.cpp](https://github.com/keyboardio/Kaleidoscope/blob/fe71904441b6f02698ac59b53efae0c159bbb772/src/kaleidoscope/hid.cpp#L23). When the `U`(`{`) key is processed, it sends an extra keyboard HID report, but that report is only partially complete, because the other held key (`C`) hasn't been handled yet, so it sends a report with `lshift`, but not `c`, erroneously telling the OS that the `c` key has been released. Then the regular report gets sent, and it has `lshift`, `[`, and `c` in it. Since the keycode for `c` comes first in the report, the output is `C{`.

I've submitted a [PR to KeyboardioHID](https://github.com/keyboardio/KeyboardioHID/pull/29) with a different way of working around the ChromeOS bug. If I remove the code for this workaround from hid.cpp, the extra characters don't appear in the output. I'll submit a PR here to fix this issue.

obra, 2018-01-23:
I very much want @algernon to chime in on this one. There's an additional thing going on here related to masking key release events so  that 
Press FN
Press U
Release Both

doesn't ever result in "{u"

That _may_ be what you're running into. 

(Please note that my caution around touching these code paths is mostly due to the fact that they're very fiddly and hard to test well. Once we have the issue well characterized, then we can figure out a good code fix more easily.)

gedankenexperimenter, 2018-01-23:
I certainly agree that more people should look at this before merging it!

Regarding the "{u" prevention: I know how that's done, and I'm sure it's not related. That isn't a matter of masking; it's done by only updating `liveCompositeKeymap` when keys toggle on, so the `{` key doesn't become `u` even after `fn` is released and the layer state gets updated — and stays that way until the next time `u` is pressed.

gedankenexperimenter, 2018-01-23:
Also, please forgive me if I've been a bit pushy about this. I'm just enthusiastic about working on it; I don't want to rush anyone into pulling changes carelessly, and this certainly isn't a major problem.

obra, 2018-01-23:
There's nothing wrong with advocating for technology improvements, as long
as everybody's friendly and respectful. we're good.

gedankenexperimenter, 2018-01-23:
I do have a tendency to send a flurry of messages, which sometimes unintentionally conveys a sense of urgency. For the record, I'm still not completely happy with my understanding of this issue.

algernon, 2018-01-24:
Assigning this to myself, so I won't forget to look at it tomorrow. I'm pretty sure this is related to keyboardio/Model01-Firmware#23, and that has an explanation that I hope is understandable. Mind you, this is just a hunch now, I've spent a total of 10 minutes reading and understanding this issue so far. Will have a closer look tomorrow.

gedankenexperimenter, 2018-01-25:
It's not the fact that those `shift` is applied to the `C`; it's the fact that the `C` appears at all that's the issue. It happens because a report gets sent without the keycode for `c` when the modifier-flagged key is pressed.

obra, 2018-01-25:
Oh! I think I might see.

gedankenexperimenter, 2018-01-25:
I plan to test this again soon, and verify that it doesn't happen with un-modifier-flagged keys that get handled before the modifier-flagged key. I'm pretty sure it happens with the `C` & `{` because the `{` gets handled first in `actOnMatrixScan()`.

gedankenexperimenter, 2018-01-25:
Indeed, it's as I expected. With the default firmware, holding down `fn` + `c` and tapping `{` (on the `U` key) produces spurious characters, whereas doing the same thing, but holding `e` instead of `c` does not. The `E` key gets handled in `actOnMatrixScan()` before the `U` key (rows are scanned from right to left, alternating between left and right halves). In fact, `E` gets handled immediately before `{`, but after `}`, so holding `fn` + `E`, and tapping `}` produces `E}`.

closed 2018-03-09 by obra