keyboardio/Kaleidoscope #285: Copyright info missing

gedankenexperimenter, 2018-01-25:
Only a few files in the repository have any copyright information attached to them, and those are all in the examples directory. The copyright holder(s) should probably decide what to do about that.

I'm guessing the following lines could apply to basically everything in this repo:
```
Copyright © 2013-2018 Jesse Vincent <jesse@keyboard.io>
Copyright © 2017,2018 Gergely Nagy <algernon@madhouse-project.org>
```
Or maybe it should just be:
```
Copyright © 2013-2018 Keyboardio, Inc. <jesse@keyboard.io>
```
…but those are just guesses, based on the logs. I don't know if others have made contributions significant enough to warrant a copyright line.

gedankenexperimenter, 2018-01-25:
On a related note, I'm curious about why GPLv2 was chosen for the license, rather than GPLv3.

obra, 2018-01-25:
I personally prefer the GPLv2 to v3. It's been a few years since I last
read v3 in depth, so my memory of the objections is a little bit fuzzier.

As to the actual copyright,

Copyright © 2013-2018 Keyboardio, Inc. <jesse@keyboard.io>

definitely belongs there. However, for work that Gergely did when he wasn't
under contract to us, it's not wrong for him to also be asserting his
copyright. I'll leave decisions on which files that touches to him, though.