keyboardio/Kaleidoscope #288: "Multiple libraries found"

simoncozens, 2018-02-15:
I'm getting an "exit status 1" running `make` in `Model01-Firmware`, and I think it's because the `kaleidoscope-builder` is getting a bit over-zealous with the library directories. Running with `VERBOSE=1` gives:

```
Building output/Model01-Firmware/Model01-Firmware (0.0.0-gv1.22-3-gb745) ...
Using board 'model01' from platform in folder: /Users/simon/Documents/Arduino/hardware/keyboardio/avr
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Detecting libraries used...
(...)
Multiple libraries were found for "Kaleidoscope.h"
 Used: /Users/simon/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidos
cope
 Not used: /Users/simon/Documents/Arduino/hardware/keyboardio/avr/libraries/Kale
idoscope/src
Multiple libraries were found for "HID.h"
 Used: /Users/simon/Documents/Arduino/hardware/keyboardio/avr/libraries/HID
 Not used: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/librarie
s/HID
```

Here's the command that's actually being run:

```
/Applications/Arduino.app/Contents/Java//arduino-builder -compile -hardware '"/Users/simon/Library/Arduino15/packages"' -hardware /Applications/Arduino.app/Contents/Java//hardware -hardware /Users/simon/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java//hardware/tools -tools /Applications/Arduino.app/Contents/Java//tools-builder -fqbn keyboardio:avr:model01 -libraries . -libraries /Users/simon/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope -libraries /Users/simon/Documents/Arduino/hardware/.. -libraries '"/Users/simon/Documents/Arduino/libraries"' -build-path /tmp/tmp.V0uMtqbxQj -ide-version 100607 -prefs 'compiler.cpp.extra_flags=-std=c++11 -Woverloaded-virtual -Wno-unused-parameter -Wno-unused-variable -Wno-ignored-qualifiers  ' -warnings all -quiet ./Model01-Firmware.ino
```

gedankenexperimenter, 2018-02-15:
The second HID.h is normal; the one from Keyboardio overrides the standard Arduino version. Have you got an extra Kaleidoscope.h file in you `libraries/Kaleidoscope` folder?

simoncozens, 2018-02-15:
Nope - the directory looks normal:

```
simon@Simons-MBP-4 ~/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope $ ls
CODE_OF_CONDUCT.md  Makefile   bin  etc       library.properties
LICENSE             README.md  doc  examples  src
simon@Simons-MBP-4 ~/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope $ git status
HEAD detached at fe71904
nothing to commit, working tree clean
```

If it's normal to see multiple libraries, does that mean the source of the "exit status 1" is elsewhere?

algernon, 2018-02-15:
It is most likely elsewhere. Multiple libraries are handled by the build tools, so that shouldn't be it.

Can you put the full build log (with `VERBOSE=1`) up somewhere?

simoncozens, 2018-02-15:
Here you go: https://gist.github.com/simoncozens/8ab246d2a4ddd29a991cc6ea9916f5b2

simoncozens, 2018-02-15:
As an additional data point, building with the Arduino GUI seems to work.

simoncozens, 2018-02-15:
OK, so I waded through all the yak hair and fixed it.

For the Googlers, if you want to build Kaleidoscope using the command line on a Mac, first run `export TMPDIR=/tmp` or the build will silently fail because the version of `ctags` shipped with the Arduino GUI falls over if `TMPDIR` is unset (see https://github.com/universal-ctags/ctags/issues/1647).

Bet you weren't expecting that.

closed 2018-02-15 by simoncozens

algernon, 2018-02-15:
Oh wow.

gedankenexperimenter, 2018-02-15:
Interesting. I've never had that problem building on macOS. I'll have to check my machine when I get a chance.

gedankenexperimenter, 2018-02-15:
I'm still wondering about the original warning you saw:
```
Multiple libraries were found for "Kaleidoscope.h"
 Used: /Users/simon/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope
 Not used: /Users/simon/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/src
```

This appears to indicate that it found a `libraries/Kaleidoscope/Kaleidoscope.h` file, which shouldn't exist.

gedankenexperimenter, 2018-02-15:
@simoncozens – What version of macOS is your machine running? `TMPDIR` _is_ normally set on mine, but it's not `/tmp`. Right now it's `/var/folders/5j/1cp97c6d49v5xwhfqsfkn5w80000gn/T`, and, in fact, overriding that is not as simple as running `export TMPDIR=/tmp` before `make`.

obra, 2018-02-15:
I'd love a patch to kaleidoscope-builder to set TMPDIR if it's unset before
build.

gedankenexperimenter, 2018-02-15:
Possibly related – I'd love a patch to kaleidoscope-builder that would allow `make; make flash` to just build everything once.

obra, 2018-02-15:
(It's worth getting ccache set up for avr-gcc if you're doing frequent
builds.)

simoncozens, 2018-02-15:
OK, it looks like it's my environment. $TMPDIR is set normally, but I'm using iTerm which does a "login -f simon" on start, and that clears $TMPDIR. I'm not entirely sure why I told iTerm to do that; maybe I hit this bug at some stage and followed the workaround: https://github.com/Homebrew/legacy-homebrew/issues/29843

I can confirm there's definitely no `Kaleidoscope.h` under `/Users/simon/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope`.

gedankenexperimenter, 2018-02-15:
Yeah, I checked that, too — I get the same duplicate `Kaleidoscope.h` warning, even though there's no duplicate.

gedankenexperimenter, 2018-02-15:
And when I set `$TMPDIR` explicitly, that has no effect on the location used during the build…