keyboardio/Kaleidoscope #293: The `}[` → `}{` bug (fast typing issue)

lydell, 2018-02-17:
See: https://community.keyboard.io/t/synthetic-modifiers-and-fast-typing-the-thing/1005

Type a `}` (fn+i) and then a `[` (fn+o). The result? `}[` as expected. (Assuming the default layout.)

Now, press and _hold_ fn+i. While holding, press fn+o. The result? `}{`! _Not_ `}[` as one would expect (since holding `a` and pressing `b` results in `ab`).

Why does this happen? Because there is no `}` key in the en-US QWERTY layout. What the firmware does is first sending `shift down` directly followed by `] down` (which results in a `}` being typed), and then `shift up` followed by `] up` when releasing the key.

So when you hold fn+i (“the `}` key”) the OS thinks that the shift key is held. When you press fn+o (“the `[` key”), the OS shifts it into `{`.

Is this a big problem? Not really, I’d say. I don’t think people type a `}` followed by a `[` very often. Because that’s when a problem _could_ arise. Many times when typing two characters quickly after each other, you actually press the second key before releasing the first. In other words, _holding_ the first key while hitting the second.

But it _is_ a problem when using the same technique for avoiding the altgr key. I type mostly in English, and that works really well. Occasionally, I type in Swedish. This requires three more letters: å, ä and ö. To get those, I use the international en-US QWERTY layout in my OS, which allows typing for example altgr+q to get an ä. But I don’t want to hold altgr when typing those letters. So I’ve made a key that’s bound to `RALT(Key_Q)`. Nice.

The problem arises when I want to write the word “är” which means “is”. A pretty common word, that is. Guess what happens when I write that quickly? I get “äë”, because altgr+r gives an ë.

Luckily, this is not a deal breaker in any way. More of an occasional annoyance. And it is totally possible to use another keyboard layout in your OS, or use some key rebinding software, to work around the issue.

@gedankenexperimenter mentioned some ideas for solving this:

> I have a solution for this in planning stages, but it involves some very extensive changes to the Kaleidoscope core.
>
> masking held keys that have a modifier & printable keycode combination when a non-modified printable key is pressed.
>
> we can check for keys with printable keycodes that have modifiers applied — those are the ones that aren’t “real” modifiers. So when a key with a printable keycode toggles on, we can check for pressed keys that meet that criterion, and mask them

gedankenexperimenter, 2018-02-17:
I have two ideas for solving this, but the good one involves major changes to Kaleidoscope that are not backwards-compatible.

The other idea is probably too impractical to be seriously considered; even if I just treat all five mod flags as a group, and don't try to deal with them individually, it turns what was four very simple lines of code in `handleKeyswitchEventDefault()` into a byzantine maze of conditionals, and I haven't even got all the way to a solution yet.

gedankenexperimenter, 2018-02-17:
>Why does this happen? Because there is no `}` key in the en-US QWERTY layout. What the firmware does is first sending `shift` down directly followed by `]` down (which results in a `}` being typed), and then shift up followed by `]` up when releasing the key.

This is not exactly true; the `shift` is added, and a report is sent, then the `]` is added to the same report, which is re-sent. Later, when the key is released, both keycodes are removed (actually, not added back int the report, which is cleared and re-populated every scan cycle) in the same report.

lydell, 2018-02-17:
I just tried to write down what I observed when using `xev` :)

noseglasses, 2018-02-19:
I played a little with my keyboard simulator [Leidokos-Python](https://github.com/CapeLeidokos/Leidokos-Python) to check the actually generated key reports.

Here is a little test script.
```python
import kaleidoscope
from kaleidoscope import *
from leidokos import *

driver = TestDriver()
driver.debug = True
driver.addPermanentReportAssertions([DumpReport()])

driver.keyDown(3, 6) # The fn key
driver.scanCycle()
driver.keyDown(1, 12) # letter 'i'
driver.scanCycle()
driver.keyDown(1, 13) # letter 'o'
driver.scanCycle()
driver.keyUp(1, 13)
driver.scanCycle()
driver.keyUp(1, 12)
driver.scanCycle()
driver.keyUp(3, 6)
driver.scanCycle()
```

The output is 
```
################################################################################

Leidokos-Python

author: noseglasses (https://github.com/noseglasses, shinynoseglasses@gmail.com)
version: 6b80915239e9d55ef5b9121801cd0b734e149208

cycle duration: 5.000000
################################################################################

0000000000 + Activating key (3, 6)
0000000000 Running single scan cycle
0000000000    Scan cycle 1
0000000000       No keyboard reports processed
0000000005 
0000000005 + Activating key (1, 12)
0000000005 Running single scan cycle
0000000005    Scan cycle 2
0000000005       Processing keyboard report 1 (1. in cycle 2)
0000000005          0 queued report assertions
0000000005          1 permanent report assertions
0000000005             Report assertion passed: Dump report: lshift 
0000000005       Processing keyboard report 2 (2. in cycle 2)
0000000005          0 queued report assertions
0000000005          1 permanent report assertions
0000000005             Report assertion passed: Dump report: lshift ]/} 
0000000005       2 keyboard reports processed
0000000010 
0000000010 + Activating key (1, 13)
0000000010 Running single scan cycle
0000000010    Scan cycle 3
0000000010       Processing keyboard report 3 (1. in cycle 3)
0000000010          0 queued report assertions
0000000010          1 permanent report assertions
0000000010             Report assertion passed: Dump report: lshift [/{ ]/} 
0000000010       1 keyboard reports processed
0000000015 
0000000015 - Releasing key (1, 13)
0000000015 Running single scan cycle
0000000015    Scan cycle 4
0000000015       Processing keyboard report 4 (1. in cycle 4)
0000000015          0 queued report assertions
[... 69 lines cut ...]
0000000000          1 permanent report assertions
0000000000             Report assertion passed: Dump report: i o 
0000000000       1 keyboard reports processed
0000000005 
0000000005 - Releasing key (1, 13)
0000000005 - Releasing key (1, 12)
0000000005 - Releasing key (3, 6)
0000000005 Running single scan cycle
0000000005    Scan cycle 2
0000000005       Processing keyboard report 2 (1. in cycle 2)
0000000005          0 queued report assertions
0000000005          1 permanent report assertions
0000000005             Report assertion passed: Dump report: none
0000000005       1 keyboard reports processed
0000000010 

################################################################################
Testing done
################################################################################

0000000010 All tests passed.
```

Leidokos-Python is nice to check corner cases that require finger twisting and good timing to generate problematic input key sequences. Its Windows version is still work in progress and is going to be
released soon. The Linux and MacOS versions are already fully functional.

gedankenexperimenter, 2018-02-19:
The more likely case is a rollover, where the two keys are released in the same order in which they're pressed, but that's certainly a good demonstration of the scan order problem.

tremby, 2018-03-05:
I've noticed that when I type ` {` (that is, space, then open-brace which I have on fn-g) too quickly (whereby I'm still actually holding space when I press the `g`) I get out *two* spaces and then the `{`. Might this be the same bug or related?

gedankenexperimenter, 2018-03-05:
I'm guessing it's the same bug. Which keys (R?C?) are your `space` and `g` keys?

tremby, 2018-03-05:
Space is in the normal position (but I've removed the binding of fn-space -> enter) of R1C8, and `g` is where the `{` is printed on the keycap (qwerty `u` -- R1C11).

gedankenexperimenter, 2018-03-06:
I'm 95% sure that means your `{` key gets scanned before `space`, which means I would expect the extra space in the output because of this bug. The rows get scanned right to left, and `space` is the last key scanned in R1.

obra, 2018-03-06:
I wonder if we should be prepopulating the matrix state with the previous
matrix state, updating each bit as we scan. Since it's pretty much
impossible for a human to hit two keys in the same scan cycle, it _should_
work identically.

gedankenexperimenter, 2018-03-06:
It's not the keyswitch matrix state that's the problem, though; it's the HID report.

obra, 2018-03-06:
Ah, the way it got described, my understanding was that the problem was due
to a scan-order issue.

gedankenexperimenter, 2018-03-06:
Right. It is and it isn't a scan order issue.

It's not because of the order the keyswitches get _scanned_ (by KeyboardioScanner), but it is because of the order the event handler hooks get called (by actOnMatrixScan()). The problem is that there's a partial HID report when an extra report is sent with the modifier only when a modified key is processed, and if there's another key (still) pressed that's handled later, its keycode(s) won't be in that report, so it will seem like they were released, and immediately pressed again (from the point of view of the HID report).

lydell, 2018-03-14:
Tried this on an Ergodox EZ today (which also has `[` and `}` in an ”fn” layer by default). It does not have this problem. `xev` says shift is released before `[` is typed. Ergodox EZ uses the [qmk](https://github.com/qmk/qmk_firmware/) firmware. Just mentioning this in case it somehow helps peeking at their code or something :)

gedankenexperimenter, 2018-03-15:
QMK may, indeed have a solution to this problem, but on Kaleidoscope, it's key position and hardware dependent. If the keys in question have their positions swapped, the behaviour on the Model01 will change. I haven't looked at the code for QMK, so I don't know if that's also the case with the their firmware.

mgsloan, 2018-03-15:
I only just started poking around in the kaleidoscope code, and I do not understand the discussion of reports above.  From my naive perspective it seems like the following is what's happening:

* The `LSHIFT` macro adds `SHIFT_HELD` to the flag to the `Key` struct in the keymap lookup table.

* In `Kaleidoscope/src/key_events.cpp`, there is a function called `handleKeyswitchEventDefault`. It gets called for most normal key presses. This invokes `pressKey` and `releaseKey` from `Kaleidoscope-HIDAdaptor-KeyboardioHID/src/kaleidoscope/hid.cpp`.

* The source for pressKey is here: https://github.com/keyboardio/Kaleidoscope-HIDAdaptor-KeyboardioHID/blob/6dc7ec5067f79e376496001624dea90763ab3e48/src/kaleidoscope/hid.cpp#L70 .  It seems to check which flags are on the key, and sends the modifier keypresses that correspond to those modifier keys.

The issue is that `pressKey` will not send any release signals for currently held modifier keys.  The only sensible solution I can think of, and I'm guessing probably what the QMK firmware does, is keep track of what modifiers the computer thinks are held down, based on info sent by the keyboard so far.  Then there would likely need to be bitflags indicating things like "SHIFT_NOT_HELD" which will cause the shift key to be released.

This gets rather tricky, because the implementation of `releaseKey` will need to know whether it should re-press the shift key or not.

algernon, 2018-03-15:
> The issue is that pressKey will not send any release signals for currently held modifier keys. The only sensible solution I can think of, and I'm guessing probably what the QMK firmware does, is [...]

We do clear the report every cycle, so if `SHIFT` is not re-pressed, then it will amount to a release event. In the report that gets sent to the host, there is no "release" event. There is only the current state.

The problem here is that a shifted key is sent in the same report as an unshifted one. Keeping track of presses/releases won't solve this in and of itself. What would solve it, is to make sure we send such shifted keys separately (which is what QMK is doing, afaik).

mgsloan, 2018-03-15:
Thanks for the explanation!  I didn't realize that the keyboard periodically reported the set of held keys.

The logic for this does indeed get rather hairy, seems like this feature would inevitably get in the way of NKRO.

closed 2018-08-11 by obra