keyboardio/Kaleidoscope #299: kaleidoscope-builder may choke on white space in paths

kingparra, 2018-02-21:
There are a lot of unquoted parameter expansions in Kaleidoscope/bin/kaleidoscope-builder. Running shellcheck on it produces a litany of errors related to word splitting. If a path used in this scrip has spaces in it, the script split the path into multiple arguments, which in turn will probably cause it to fail. I'd like to humbly suggest that you run shellcheck against this file and correct the quoting problems, and change the shabang to `#!/usr/bin/env bash`, since the script already uses bashisms and bash can be found on both mac and Linux. (You don't gain any compatibility by invoking this script through `sh`.) I would re-write it, but am quite the noob and don't want to break anything.

kingparra, 2018-02-21:
Just an update. I'm working on a pull request related to this issue, and will link to it once it's up. Although, no promises -- I'm still learning how to use git.

algernon, 2018-03-09:
Fixed by #305.

closed 2018-03-09 by algernon