keyboardio/Kaleidoscope #303: Support for non-US languages & layouts

algernon, 2018-02-28:
At the moment the key definitions are that of the USB standard, that match US QWERTY. For other languages, or other layouts, there are no easy way to match the US QWERTY keys to the internationalized ones. As there are many non-US users, we should make it a lot easier to create layouts in their own language.

For this to work, I believe that we should start off with the assumption that the OS-side layout is set to a localized one. Because traditional keyboards are set up that way, us doing so is also reasonable.

algernon, 2018-02-28:
My proposal would be this: create a `Kaleidoscope-Languages` plugin, where we'd collect the language & layout-specific key definitions. To avoid conflicts, and to allow us to have short(er) key names, the main header would not include the others, and it would be up for the end-user to include the language & layout header they want to use.

So we'd have `Kaleidoscope-Languages.h`, `kaleidoscope/lang/de-qwertz.h`, `kaleidoscope/lang/hu-qwertz.h`, and so on, with keys defined as `Key_DE_something`, `Key_HU_something`, etc. For keys that are the same as on US QWERTY, we could still have a `Key_DE_A`, for sake of being complete.

The plugin would also come with a number of examples, to get people started: they could copy the layers from there to their own sketches, and re-arrange them as they see fit.

Why not put this in core, one may ask? Because there are many languages, even more layouts, and some of them conflict. There is no technical reason we can't put it in core: they won't take up space in the compiled firmware if not used, because these are headers only; we could have the same directory layout too.

The main reason for my push to put this into a plugin is to be able to provide many examples, in a discoverable way. And because language support has a good chance of growing big. I think it'd be easier to manage as a plugin (technically not a plugin, but a header-only library, but lets not split hairs!).

obra, 2018-03-01:
I can buy that rationale.

algernon, 2018-03-01:
Another reason for pushing this to a plugin (as highlighted by @gedankenexperimenter): it makes the history more straightforward, merges easier, and so on. Because there is *only* language stuff there, no silly code and whatnot that complicate things.

jordihs, 2018-04-24:
Well, just in case this helps, here is the special key mappings that I made for my language specific layout, which is es_ES (spanish-Spain): 
``/*
   Spanish layout keys mapped to HID correspondants in english, for ease of use
*/

// `^[
#define Key_Circunflejo Key_LeftBracket
// ´¨{
#define Key_Acento      Key_Quote
// <>
#define Key_MayorMenor  Key_NonUsBackslashAndPipe
// ºª forward slash
#define Key_Ordinal_ao  Key_Backtick
// ¡¿
#define Key_Apertura    Key_Equals
// +*]
#define Key_Suma        Key_RightBracket
// çÇ}
#define Key_Cedilla     Key_Backslash
// '?
#define Key_Apostrofe   Key_Minus
// ñ
#define Key_Enye        Key_Semicolon
// -_
#define Key_Menos       Key_Slash

noseglasses, 2018-04-24:
Wouldn't it be good to add new key alias as `constexpr` in appropriate namespaces instead of adding more pre-processor macros that will increase the likelihood of naming conflicts?

Something like
```cpp
namespace kaleidoscope {
namespace key_alias {
namespace spanish {

constexpr Key Key_Circunflejo = Key_LeftBracket;
//...
}
}
}
```

There is still a dangling PR https://github.com/keyboardio/Kaleidoscope/pull/259 that is partially (at the current end of the discussion) about converting the existing macro defined `Key` constants to `constexpr`.

Porque a veces escribo in español, me gustan mucho los nuevos definiciones the teclas españoles.

jordihs, 2018-04-24:
I defined these in my .ino file and I have no background in C++ (or was it C#? Plain C? I'm at a loss...) 
So any and all corrections from the true experts are welcome. 
Glad you like them! Yo también escribo en español a veces :-P

noseglasses, 2018-04-24:
My comment was mostly directed to the core developers. I should have marked it as such.