keyboardio/Kaleidoscope #307: Is there a way to make left and right function key only switch that half of device?

LiamKarlMitchell, 2018-03-16:
Was trying to play a simple game with cursor keys and z and x and cant press them whilst using cursor.
maybe a way to remap to a game mode. 

ill do some more reading and try to flash firmware soon

cursor keys separate would have been very useful.

gedankenexperimenter, 2018-03-16:
You could have two layer — one with a transparent left side, on with a transparent right side. Then, on the base layer, one palm key is mapped to `ShiftToLayer(FN_LEFT)` and one to `ShiftToLayer(FN_RIGHT)`.

LiamKarlMitchell, 2018-03-17:
Great replies thanks.

algernon, 2018-09-28:
As there were a few useful answers provided, I think we can consider this question answered. I'm closing the issue, but feel free to reopen it (or open a new one) if you still have questions, or need help with setting up your sketch the way you want.

closed 2018-09-28 by algernon