keyboardio/Kaleidoscope #310: Adding CLI build support on Windows with Cygwin

cyrossignol, 2018-04-15:
I made some minor modifications to the firmware build process to enable building and uploading from the command-line on Windows with Cygwin so I can avoid opening that thing they call an IDE. I want to check first if we're interested in adding this support to Kaleidoscope before doing the work for more in-depth testing/validation, documentation, and opening a PR. 

For those unfamiliar: [Cygwin](https://cygwin.org) is a best-effort, POSIX-compliant project that provides a Unix-like environment on Windows and includes ports of many programs and libraries typically found on Linux/Unix/BSDs. 

Because the build system is composed of shell script and Makefile "glue", and we defer to the platform-specific distribution of Arduino for compilation and device IO, we can execute builds in Cygwin with relatively minor changes to some scripts. Most of the changes center around converting Unix-style paths if needed and using a Windows-native routine for port detection. 

Alternatively, I can port the build system to PowerShell for a purely-Windows build process if we don't mind maintaining two versions of the build logic. We could also set this up for [WSL](https://en.wikipedia.org/wiki/Windows_Subsystem_for_Linux). I guess I'm looking for some feedback about how we want to support end-to-end command-line builds on Windows, or whether this is even a goal.

obra, 2018-04-16:
> Personally, I think that Cygwin support makes sense, so does WSL.
> Reimplementing the build stuff in PowerShell is not something I'd recommend.

Indeed, my goal is for us to have one implementation and as much
cross-platform compatibility as possible.

The reason the commandline Windows support is poor has nothing to do with
intent and everything to do with experience (or lack thereof ;)

I'd love to have something that works on Windows without cygwin or WSL, but
short of switching to a compiled language or powershell, I don't know what
our options would be.

-j

cyrossignol, 2018-04-17:
Thanks for the replies. There's really only one option if we can only maintain the current implementation: Windows users will need a shell script interpreter to do a seamless command-line build. 

I don't know the OS distribution of Keyboardio users... if most use Mac/Linux, it may be fine to say, "if you're on Windows, use the IDE, or follow these docs to set up all the not-really-supported tools needed to build with the command-line."

To make Windows support a bit more official, we could also distribute a [BusyBox executable built for Windows](https://frippery.org/busybox/) (maybe with keyboardio/Kaleidoscope-Build-Tools) and possibly GNU Make from MinGW (the binaries are both pretty small, and I think the licenses are compatible). This saves users from the need to install tooling themselves, but it may add a bit of bloat to the project. 

Sounds like we want to avoid porting to PowerShell for the time being. It may provide a secondary cross-platform build option for the future, though&mdash;the open-source PowerShell Core recently released stable versions for Mac and Linux.

Personally, I live in Cygwin when I'm on Windows, but I don't mind setting up one of the other options if the demand is there and the project is open to them.

obra, 2018-04-18:
> though—the open-source PowerShell Core recently released stable versions

I feel like telling folks who want a commandline dev environment to use
cygwin or WSL isn't that wrong. But maybe other windows users will chime in
and tell us they really want something else.

(Long story short: happy to take cygwin portability patches)

gedankenexperimenter, 2018-04-18:
Back when I had to use Windows machines, one of the first things I always did was install Cygwin. To me, that was the only way to make it (marginally) usable. The only problem is that it's somewhat complex to install, unless the only packages needed are selected by default.

cyrossignol, 2018-04-18:
The "base" package category seems to contain all the typical programs used by the build scripts with the exception of Make (and maybe Python for the code quality stuff). Because we don't really need any deeper OS integration for Kaleidoscope, new Cygwin users can probably get by with the help of some light documentation. 

By using the setup program's command-line interface, we might document a command or provide a script that someone can run to install the minimum environment needed to build the firmware. It's pretty stable, surprisingly&mdash;I keep a similar script with my dotfiles that has worked for years without any issues. 

Seems obvious that this should also work for Git Bash/MSYS...if someone runs a build from the command line, they likely cloned the repo using Git for Windows. This may actually be a better default recommendation for Windows users. 

Moving forward with these two...looking for a more complete set of things that need testing. I'm not very familiar with the code base, and I've only looked at making the main build/upload work. Is there anything special needed for plugins? Or are those Makefiles mostly for CI?

Speaking of CI...it may be worth getting Windows builds running on something like AppVeyor while we're at it.

obra, 2018-04-20:
> surprisingly—I keep a similar script with my dotfiles that has worked for

Plugin makefiles are mostly for CI. I'd be happy to have something testing
builds on Windows and/or MacOS, though it -should- be just to test the
build infrastructure, since the outputs should be the same.

Best,
Jesse

benze, 2019-09-14:
@cyrossignol Are you able to build successfully in Cygwin?  I've been struggling for the last couple of days trying to get this built and deployed on a Windows box.  Building in Cygwin fails miserably at the moment.  I can build in WSL, but I can't flash the device (even though it is listed on /dev/ttyS4 in WSL, WSL fails to communicate with it), and when I try in a Virtual Box VM, it fails to go into bootloader mode and reconnect to the VM.

So I am completely stuck at the moment and can't get any traction in any OS.

Do you have a fork somewhere with your build script changes that will allow me to build/Flash from Cygwin?

cyrossignol, 2019-09-14:
Hi @benze, 

Unfortunately, I may not be of much assistance at the moment. Shortly after I opened this ticket, life kicked-in and I haven't had time to finish the Cygwin integration that I wrote about. I'm sure the firmware and build tooling changed quite a bit in the year-and-a-half since I last flashed my keyboard--I'll need to find that old code and re-familiarize myself with this project. 

Besides Cygwin, I remember successfully flashing the keyboard without a hitch from Linux in a VirtualBox VM on a Windows host. It required adding a device filter under _Settings_ &rarr; _USB_ for the keyboard to enable direct communication with the bootloader. This may be a path forward in the meantime. 

I'd love to finish what I started. I'll try to find some time over the next few weeks to pick up where I left off, or at least to find my changes and push them up so that someone can rebase those if needed.

benze, 2019-09-15:
@cyrossignol If you can push the changes into a repo somewhere, it might give me a little direction.  I've been trying to hack around the build projects, but even if I get this working, it's pretty much guaranteed to be non-portable to someone else, which I don't want to do.  If you actually got something up and running cleanly, it might give me a better idea/indication of what directin to go.

I've tried added the USB device filters, filtering on the DeviceId 2000 and 2001 and the VendorId 1209, but it only seems to pick up the keyboard when it is in non-bootloader mode (id: 2001).

If you got it to work though, it does give me some hope and I'll keep pushing along that path with the VB VM.

obra, 2020-06-08:
Closing, as I believe #703 resolves this.

closed 2020-06-08 by obra