keyboardio/Kaleidoscope #312: Setup instructions missing a step for Arduino IDE

goodmami, 2018-04-24:
I'm new to Arduino, so please forgive any incorrect assumptions and general ignorance.

I was following the setup instructions for the first time (namely [Install-Arduino-support-on-Linux](https://github.com/keyboardio/Kaleidoscope/wiki/Install-Arduino-support-on-Linux) and [Edit-A-Keymap](https://github.com/keyboardio/Kaleidoscope/wiki/Edit-A-Keymap)), and it gave the following instruction:

> Open the Arduino IDE. In the “File” menu, click on the “Examples” submenu. Scroll down to ‘Model01-Firmware’.

But `Model01-Firmware` was not in the Examples list. Once I changed the "board" (not sure what this means... the target hardware profile, I guess?) in Tools>Board to `Keyboardio Model 01`, then the `Model01-Firmware` option appeared in the Examples submenu.

Is this the preferred method of getting to that menu? If so, I think the board-changing step should be added to the wiki.

goodmami, 2018-04-25:
*Update*: I had forgotten a step in the README, which was to clone the Model01-Firware repo, but cloning the repository didn't seem to help the original issue (I still had to select the "Keyboardio Model 01" board in the IDE). Maybe the step to clone that `Model01-Firmware` repo should be included or linked from the wiki, as currently the complete instructions are spread across several documents. I think I was originally using the firmware repo that was pulled in from the `git clone --recursive https://github.com/keyboardio/Arduino-Boards.git` command.

On rereading the README, however, I noticed that using the Arduino IDE was not necessary, so I could just edit Model01-Firmware.ino with `vim` (or some other `$EDITOR`) and then `make flash`. This was much easier.

per1234, 2018-04-25:
I have made an edit to the "Install Arduino support on Linux" wiki page that resolves this issue:
https://github.com/keyboardio/Kaleidoscope/wiki/Install-Arduino-support-on-Linux/_compare/8558887fbd48b72c0d23c5c58b354eb8e6ff69a5...e20ad9fa07629a41ad7a5b9bd1e18c7aa2e3e89f

Examples bundled with a hardware package are only displayed in the Arduino IDE's **File > Examples** menu when a board of that hardware package is selected from the **Tools > Board** menu.

The Windows and Mac instructions already had a step to select the correct board but that was missing from the Linux instructions.

goodmami, 2018-04-25:
@per1234 excellent, this sounds like a good fix. Thanks!

algernon, 2018-04-25:
Thanks for both the report, and the fix! <3

closed 2018-04-25 by algernon