keyboardio/Kaleidoscope #323: Adding a new hook from a plugin

algernon, 2018-05-16:
At the moment, it is not easily possible to add a new hook entirely from a plugin: all the helpers will try to put symbols under `kaleidoscope::Hooks`, and use methods from `kaleidoscope_internal::EventDispatcher`. These are not extensible from outside, so if a plugin wants to add a hook, it has to duplicate all the magic macros.

It would be nice if they had an extra argument, letting the caller control where the symbols end up at.

algernon, 2018-05-17:
Another option that would help - and might even be better - is to turn `kaleidoscope::Hooks` from a class to a namespace. That allows plugins to extend the namespace, and reuse the same macros we use in core.

The downside is that hooks will now be callable from anywhere - which may not be wrong, since we would like to allow plugins to have their own hooks.

algernon, 2018-05-17:
The main motivation for the namespace conversion was to allow plugin-defined hooks. But since those will need a separate... ehh.. lets call it infrastructure?... the conversion is pointless.

Thanks a lot for the ideas, by the way!

Closing this PR, because I'll do it in a very different way.

noseglasses, 2018-05-17:
It just occured to me that there is at least one issue that needs to be solved, which is the friendship of the `Hook` class. If we would provide a macro that would define siblings of the `Hook` class in user space (I'll call it that way to distinguish from core-space - hello Linux kernel!), we would need to pass this macro the name(s) of those classes and functions that are meant to be allowed to call a specific set of user space hooks.

Example:
```cpp
#define _FRIEND_DECLARATION(CLASS_OR_FUNCTION) \
   friend CLASS_OR_FUNCTION;

#define UNWRAP(...) __VA_ARGS__

#define KALEIDOSCOPE_DEFINE_HOOK_SYSTEM(HOOK_SYSTEM, EVENT_HANDLER_DEFINITION, FRIENDS)  \
   namespace kaleidoscope { \
      class HOOK_SYSTEM {  \
         public:  \
            MAP(_FRIEND_DECLARATION, UNWRAP FRIENDS)  \
         private:
            EVENT_HANDLER_DEFINITION(DEFINE_WEAK_HOOK_FUNCTION)
      };
   }

...
KALEIDOSCOPE_DEFINE_HOOK_SYSTEM(
   FancyPluginsHooks, // The user-space hook system's name
   FANCY_PLUGINS_FOR_EACH_EVENT_HANDLER, // The plugin's version of _FOR_EACH_EVENT_HANDLER (defined somewhere)
   ( // classes and functions (fully qualified names!) that are supposed to allowed to call hooks of 
     // hooking system PluginsFancyHooks
      class Plugin, 
      void anyArbitraryFuntion()
   )
)
```

A modified version of `INSTANTIATE_WEAK_HOOK_FUNCTION`, called with `FANCY_PLUGINS_FOR_EACH_EVENT_HANDLER` would take care of the instantiation of the hook system.

noseglasses, 2018-05-17:
```cpp
// Plugin defines own hooks
//
#define FANCY_PLUGINS_FOR_EACH_EVENT_HANDLER(HOOK_SYSTEM, OPERATION, ...) __NL__ \
                                                                          __NL__ \
   OPERATION(HOOK_SYSTEM, onWhatever,                                     __NL__ \
               _NOT_ABORTABLE,                                            __NL__ \
               (),(), ##__VA_ARGS__) 

// Modified version (part of core) - note the added HOOK_SYSTEM parameter
// that replaces the name Hook in the macro's implementation
//
#define _REGISTER_EVENT_HANDLER(                                                 \
HOOK_SYSTEM, HOOK_NAME, SHOULD_ABORT_ON_CONSUMED_EVENT, SIGNATURE, ARGS_LIST) __NL__ \
   ...

namespace kaleidoscope {
   FANCY_PLUGINS_FOR_EACH_EVENT_HANDLER(FancyPluginsHooks, _REGISTER_EVENT_HANDLER)
}
```

Note: I even added `HOOK_SYSTEM` as parameter to macro `FANCY_PLUGINS_FOR_EACH_EVENT_HANDLER` to enable definition of several hook systems  with different names but the same hooks.

Of course `DEFINE_WEAK_HOOK_FUNCTION` and the other similar macros define by the core would also need to be modified to obtain the `HOOK_SYSTEM` as their first parameter.

noseglasses, 2018-05-17:
... I called it `HOOK_SYSTEM`, might as well be named `EVENT_HANDLER_SYSTEM`.

The proposed changes would enable a plugin to provide additional event handlers that could be used in derived plugin classes. It would require instances of specific plugins to be registered with the plugin's hooking/event handling system from within the sketch, just like `KALEIDOSCOPE_INIT_PLUGINS` does for the default/core hooks/event handlers. 

This would require an additional macro, similar to `KALEIDOSCOPE_INIT_PLUGINS` that would be invoked on the subset of plugin instances that support the additional event handlers. A wrapper for `KALEIDOSCOPE_INIT_PLUGINS` wouldn't do, because it couldn't distinguish, which plugins objects support which type of handlers, i.e. which plugin object is of a type that is derived from a plugin class that defines additional hooks.

algernon, 2018-11-24:
I think this is much less important now that most of the plugins are in the same repo. Adding a new hookpoint is quite cheap, so the only motivation for adding new points via another plugin would be that the hook point is closer to the plugin. But we already "broke" that with `onFocusEvent`, and thus we have a hook point in core that is only of use when using a specific plugin, because the call site is there.

We can do the same for other plugins, would they need a hookpoint. So... I'd say we can close this as wontfix for now.

commit 7a53abe2 (2018-05-17): Turn kaleidoscope::Hooks into a namespace

We want to allow plugins to have their own hooks, without having to touch the
core firmware. For this reason, protecting the hooks via a class and friend
declarations does not make sense. Nor does a class, for that matter, as it is
not extensible from the outside. A namespace, however, is.

And this is what this change does: turns `kaleidoscope::Hooks` into a namespace
called `kaleidoscope::hooks`. This way, plugins can declare new hooks within it,
and don't need to sync with core.

Fixes #323.
commit 74382ace (2018-05-17): Turn kaleidoscope::Hooks into a namespace

We want to allow plugins to have their own hooks, without having to touch the
core firmware. For this reason, protecting the hooks via a class and friend
declarations does not make sense. Nor does a class, for that matter, as it is
not extensible from the outside. A namespace, however, is.

And this is what this change does: turns `kaleidoscope::Hooks` into a namespace
called `kaleidoscope::hooks`. This way, plugins can declare new hooks within it,
and don't need to sync with core.

Fixes #323.
closed 2018-12-04 by obra