keyboardio/Kaleidoscope #329: Possible regression in new API

EJamesW, 2018-05-21:
[Model01-Firmware.ino.txt](https://github.com/keyboardio/Kaleidoscope/files/2022072/Model01-Firmware.ino.txt)

After pulling the latest code on May 16th at 12:32 (CEST), and making the requisite changes to my sketch in order to use the new API correctly, there were some problems hitherto not encountered:

- two macros stopped working: <code>MACRO_TOGGLE_QUKEYS</code>, and <code>MACRO_TMUX</code>.
In the case of <code>MACRO_TMUX</code>, <code>xev</code> indicated the macro was firing twice, i.e., <kbd>Down(Ctrl), Down(B), Up(B), Up(Ctrl), Down(Ctrl), Down(B), Up(B), Up(Ctrl)</kbd> 
- repeated keystrokes, in particular the <kbd>d</kbd> and <kbd>k</kbd> which are both defined as Qukeys with an alternate keycode of a OneShot <kbd>Shift</kbd> modifier. This would be inconsistent and hard to reliably reproduce. I noticed it most often when resuming typing after a pause.
- the above mentioned OneShot modifiers would sometimes not register at all even when they should have.
- OneShot shift modifiers would often lock, even though I'd set <code>OneShot.double_tap_sticky = false;</code> Actually, this only occurs when I set <code>Qukeys.setReleaseDelay(20)</code>, and with the default of <code>0</code> there's no problem with this - and furthermore it happens even when using the old API too, so this is really a separate issue.

Other possible pertinent information is that I'm using Arch Linux, I was compiling from the command line, and though all avr related packages installed via pacman are up to date, I had to keep from upgrading <code>avr-gcc</code> to version 8.1.0-1, as I could no longer compile my sketch, so it's still on version 7.3.0-1

algernon, 2018-06-17:
DUH. So obvious in hindsight. Got the problem nailed down, working on a fix. The same issue might affect other plugins too.

The root of this is the compatibility layer, and one particular assumption it made: it assumed that plugins that have a non-empty `begin` method will *always* be old-style plugins. For this reason, the default `onSetup` hook will call `->begin` (which is a no-op for v2 plugins). This in turn allows the end-user to use `KALEIDOSCOPE_INIT_PLUGINS` with old-style plugins as well.

However, this presents a problem when we have v2 plugins that implement a compatibility layer: because now their hooks will run twice! Once called by the v2 layer, once by the v1 compatibility stuff.

There are two solutions here:
 - Implement `onSetup()` for all the v2 plugins, so that it overrides the default that calls `->begin`.
 - Remove the ability to initialize v1 plugins with `KALEIDOSCOPE_INIT_PLUGINS`.

As we want to remove the compatibility layer *soon*, I'd go with the second option.

Until we figure out how to handle the case, those affected can comment out [this line](https://github.com/keyboardio/Kaleidoscope/blob/88c1d3d7e11991ce29a5189a644089763ac5d266/src/plugin.h#L86) in `Kaleidoscope/src/plugin.h` locally, for the time being.

algernon, 2018-06-17:
Another option is to disable the V1 API, which can be done when using the CLI: add the following line to either `~/.kaleidoscope-builder.conf`, or `.kaleidoscope-builder.conf` in the root directory of your sketch (usually the same as the sketch itself):

```sh
LOCAL_CFLAGS="-DKALEIDOSCOPE_ENABLE_V1_PLUGIN_API=0"
```

algernon, 2018-06-17:
I ended up changing all affected plugins, because that was the safer route. Updating the plugins should fix all these issues. Sorry it took so long!

Can I ask you to verify if the issues are fixed?

EJamesW, 2018-06-17:
I've updated all the plugins, but I'm not sure whether the changes you've made to the affected plugins meant that I shouldn't need to still add the line to `~/.kaleidoscope-builder.conf`?

In any case, I can happily say that adding that line to `~/.kaleidoscope-builder.conf` to disable the V1 API did indeed work and all the issues are fixed as far as I can tell.

algernon, 2018-06-17:
With the recent changes, adding `LOCAL_CFLAGS` to `~/.kaleidoscope-builder.conf` is not necessary anymore.

Can you test if they work without the line, too? (as far as I could tell, they do, but "works on my computer" is not good enough :P)

EJamesW, 2018-06-17:
Yep, it works now without the line.

Thank you, algernon!

algernon, 2018-06-17:
Awesome. So happy to hear that! Closing the issue then.

Thanks for your patience, and for testing the fix too!

closed 2018-06-17 by algernon