keyboardio/Kaleidoscope #342: Improve version number parsing

a-dma, 2018-07-12:
Hi,

I'd like to version the changes I do to my own firmware and I noticed there is some support for this by parsing the `library.property` file when calling `kaleidoscope-builder`. However, the script assumes a relative path that, at least in my case, is incorrect: `../../library.properties`.

I've made a couple of changes allowing to pass in the path where the properties files is located using a variable called `LIB_PROPERTIES_PATH`.  The option can be set directly in `.kaleidoscope-builder.conf`. I thought that the change was so small that it didn't warrant a PR. I'm attaching a diff to this issue.

[diff.txt](https://github.com/keyboardio/Kaleidoscope/files/2188851/diff.txt)

It would be nice if this got picked up upstream.

algernon, 2018-07-17:
The patch looks good to me, thank you! I'll try to find some time to integrate it in the next day or two.

(And for future reference, feel free to open a PR even for a tiny typo change. PRs are cheap =)

a-dma, 2018-07-17:
I was just being lazy, but since you asked... here's #343 :)

obra, 2018-07-18:
The way we populate the build information for Kaleidoscope is usually based on git tags rather than Arduino's library versions. (They're mostly useful for publishing libraries to the Arduino ecosystem.) 

I've never managed to get a build to -use- the information in library.properties. Have you? I'd love to know how!

Best,
Jesse

a-dma, 2018-07-18:
I haven't studied the build system much to know all the external information that you are adding already or would like to add in the future, but here is what I've done in brief. I've reorganized the build process a bit because I found it extremely confusing and I didn't like having to have stuff in `~/Arduino`.
After doing that and adding the changes in the Pull Request above I simply added the following to my top level `.kaleidoscope-builder.conf` file

```
GIT_VERSION="$(git describe --abbrev=4 --dirty --always)"
LIB_PROPERTIES_PATH="$PWD"
LIB_VERSION="$((grep version= ${LIB_PROPERTIES_PATH}/library.properties 2>/dev/null || echo version=0.0.0) | cut -d= -f2)-${GIT_VERSION}"

LOCAL_CFLAGS="-DKALEIDOSCOPE_HIDADAPTOR_DISABLE_KEYBOARD_BOOT_PROTOCOL=1 -DKALEIDOSCOPE_ENABLE_V1_PLUGIN_API=0"
LOCAL_CFLAGS="$LOCAL_CFLAGS -DBUILD_INFORMATION=\"${LIB_VERSION}\""
BOARD_HARDWARE_PATH="${SOURCEDIR}/hardware"
DEFAULT_SKETCH="firmware"
OUTPUT_PATH="output
```

This sets up the output path and defines `BUILD_INFORMATION` to be the version defined in library.properties and the current git hash. As I'm sure you're aware, that info can be printed by pressing `Any` in `Num Lock` mode. For example here is mine `Keyboardio Model 01 - Kaleidoscope 0.1.0-2952`.

obra, 2020-06-08:
Closing, as #343 was merged a whiel ago.

closed 2020-06-08 by obra