keyboardio/Kaleidoscope #369: Macros: Record Macros at runtime

algernon, 2018-10-16:
Would it be possible to record macros at runtime?

From the example in the README.md it looks like all macros need to be compiled. Software like vim lets you record macros on the fly for temporary use. This would be handy for a physical device. You could hit a button to record, type a few things, hit a button to stop the recording. Then, you can hit a button to play back the macro.

Is this possible?

algernon, 2018-10-16:
Originally by @theSloopJohnB, as keyboardio/Kaleidoscope-Macros#13

wxcafe, 2019-05-17:
Isn't this solved by https://github.com/cdisselkoen/Kaleidoscope-MacrosOnTheFly, as mentionned in the original thread?

algernon, 2020-01-06:
That, and the DynamicMacros plugin pretty much cover all bases here, as far as the firmware is concerned. Mind you, to close the issue, we should probably pull MacrosOnTheFly into Kaleidoscope proper.

@cdisselkoen, @obra, what do you think about that?

cdisselkoen, 2020-01-06:
Sounds good to me, if there's interest.

obra, 2020-01-06:
works for me