keyboardio/Kaleidoscope #370: Macros: EEPROM storage for the macros

algernon, 2018-10-16:
Similar how keys can be looked up from either `PROGMEM` or `EEPROM`, macros should be possible to store there, too.

We have two options for storage:

- Fixed size macros: easy to implement and work with, but wasteful.
- Dynamic sized macros: Harder to implement, and requires lots of moving around (which can wear out the EEPROM faster), but much less wasteful.

At first, lets go the easy route: we'll let the user say how many bytes they want allocated, and we'll divide that up into equal parts (using a hard-coded macro size for now).

SRAM use should be kept to a minimum, so updating a macro should not require a pre-assembled `macro_t`.

algernon, 2018-10-16:
Originally by @algernon, as keyboardio/Kaleidoscope-Macros#3

algernon, 2019-09-10:
This is something the folks at @dygmalab are very interested in, and we had a short discussion with @mattvenn about the topic. I'd like to present my proposal here, with a bit of explanation to shed some light on why I think this is a good approach.

The gist of the idea is that we'd push logic to the configurator software (in the majority of cases, that will be Chrysalis), and the storage would be nothing else but a continous `MACRO_ACTION_END`-delimited series of macro definitions, using the same step definitions as the `Macros` plugin already does. To make lookups easier at runtime, we'd build an index at init time, and every time the macros are updated via Focus. That'd be about 64 bytes of RAM used (32 macros, 2-byte offsets for each, so we can have at most 16k storage for macros initially, which is plenty). We'd also have a new range for these macros, so one can use these and built-in ones together. Whenever we want to update macros, the configurator will re-serialize _all_ of the eeprom-stored macros, and send the whole batch over to the keyboard, overwriting the existing ones. Similarly, when the configurator wants to retrieve macros, it would fetch them in one command, and parse them similar to how the firmware would.

There are a number of advantages of this approach: It's fairly simple on the firmware side: reserve storage, scan when required, do lookups, execute reusing pretty much the same functions we already have. It doesn't require us to rearrange the macros from within the firmware, nor does it require us to reserve N bytes for _each_ macro (which is potentially wasteful when using short macros, and prevents one from using longer ones) - therefore, this is more flexible.

The downside is that we'll be overwriting EEPROM fairly often. Mind you, we'd do that with in-firmware rearrangement too, so this isn't a huge disadvantage either. We wouldn't do this if we had fixed-size macros, but those aren't flexible, and are wasteful, so I'd like to avoid them. From a user point of view, dynamically sized macros are much more useful.

Another downside is that the configurator is not able to get/set a single macro, only the whole thing together. But that's how keymap and colormap work too, so we're just doing things the same way other plugins do, and we won't be pushing much more data, either.

As for the focus API, I'd propose these:

- `macros.custom`: Mirroring `keymap.custom`, to be used to get/set the macros. Returns a 0 (`MACRO_ACTION_END`) delimited set. To avoid confusion, we should signal the end of _all_ macros with double zeroes. This should be delegated to the configurator, however: when setting macros, just append a zero at the end. This saves a bit of code on the firmware side. (I can be convinced to enforce this on the firmware side too, mind you.)
- `macros.trigger`: Would run the specified macro. Useful for testing without having to assign it to a key.

That's about it. It's simply a dump of the already existing macro definitions into EEPROM, a pair of Focus commands, and some code to look things up and build an index. We can - hopefully - reuse a fair amount of interpretation code from the existing `Macros` plugin.

algernon, 2019-09-11:
A few examples for the Focus API:

```
macros.custom 8 11 8 8 8 15 8 15 8 18 0 0
```

This sets the storage to contain a single macro, which will input "hello":

- `8 11` is `Tc(Key_H)`
- `8 8` is `Tc(Key_E)`
- `8 15 ` is `Tc(Key_L)`
- `8 18` is `Tc(Key_O)`
- `0` is `MACRO_ACTION_END`

And the second zero tells the firmware that this is where macros end, the rest of the reserved space for macros is unused.

Doing a `macros.custom` without arguments would echo back the same sequence, to be parsed as shown above.

`macros.trigger 0` would tigger the above macro, and ask the firmware to run it.

algernon, 2019-09-11:
Come to think of it, we can compress these kinds of macros a _lot_, if we add an action that says "Tap everything up till the next 0". We could call that `MACRO_ACTION_TAP_KEY_SEQUENCE`, and `MACRO_ACTION_TAP_KEYCODE_SEQUENCE` (the first would use 16-bit keycodes, the second 8-bit ones).

With this, the above `8 11 8 8 8 15 8 15 8 18 0 0` sequence could be rewritten as `13 11 8 15 15 18 0 0 0`, three bytes saved. The longer the text, the more we save. The three zeroes at the end are: the first terminates the action, the second the macro, the third tells the firmware that we have no more macros.

algernon, 2019-09-11:
The hard part of this whole thing would be the configurator. Because to make best use of the available macro actions, the configurator would need to optimize the sequence a bit. Thankfully, that's not a strictly required feature, so the initial implementation can coast by without any optimization and just using down/up/tap actions. Or expose all of the actions, and let the end-user set up the macro directly.

mattvenn, 2019-09-11:
looks good. I suggest we leave the optimisation for now to make the initial implementation in the configurator easier.

mattvenn, 2019-09-11:
Also, here is the header file that defines the macro enums: https://github.com/keyboardio/Kaleidoscope-Macros/blob/master/src/MacroSteps.h

obra, 2019-09-11:
What's the maximum number of different codepoints we can have in a macro?
If we have a bit free (which we may not), we could store the stop bit in
the last step, rather than burning a whole byte on it.

Similarly, if we have at least 3 -entries- free, "0", could be "action
done, macro done, no more macros", 1 could be "action done, macro done" and
"3" could be "action done"
ᐧ

algernon, 2019-09-11:
> What's the maximum number of different codepoints we can have in a macro?

For the 16-bit sequence, our current highest codepoint is around 60k, so we could use 65533-65535 as stop signs. For the 8-bit sequences, 0xE7 (right gui) is our highest supported codepoint, so we can use 253-255 for the stop signs. This could save a few bytes, indeed.

In other words, we can't use a stop bit, but we can pack the various stop possibilities into either 16 or 8 bits, depending on whether we use 8 or 16-bit arguments for the particular step.

mattvenn, 2019-09-16:
tested on Raise - works as expected

commit 181b5df2 (2019-09-16): Dynamic Macros plugin

This implements a new plugin for Dynamic (EEPROM-stored) macros. Unlike the
Macros plugin, these macros are stored in EEPROM, and can't run custom code,
only the steps outlined in the Macros documentation.

The plugin provides two Focus commands (`macros.map` and `macros.trigger`) to
get or set the dynamic macros, and to trigger one without having to place them
on the keymap.

Fixes #370.
commit 357c0c4b (2019-10-05): Dynamic Macros plugin

This implements a new plugin for Dynamic (EEPROM-stored) macros. Unlike the
Macros plugin, these macros are stored in EEPROM, and can't run custom code,
only the steps outlined in the Macros documentation.

The plugin provides two Focus commands (`macros.map` and `macros.trigger`) to
get or set the dynamic macros, and to trigger one without having to place them
on the keymap.

Fixes #370.
closed 2019-10-07 by obra