keyboardio/Kaleidoscope #398: Qukeys: Consider making LockLayer qukeys

algernon, 2018-10-18:
Originally by @gedankenexperimenter, as keyboardio/Kaleidoscope-Qukeys#22.

gedankenexperimenter, 2020-10-03:
@algernon, @obra — What happened to the old plugin repositories? I was trying to confirm if this issue was what I think it is, so that it could be closed, but that repo seems to be gone completely now. I don't know if I have a copy of it anywhere, and for historical reasons, I'd like to be able to refer to my early work on it.

obra, 2020-10-03:
After a year, we deleted the archived repositories - they were causing
significant end-user confusion since GitHub doesn't have a way to hide
archived repositories.

The full content of that ticket before migration was:

[image: image.png]

obra, 2020-10-03:
Apparently, github strips images from mail. 

[image]

gedankenexperimenter, 2020-10-03:
Ah, oh well.
Anyway, about this issue — it is exactly what I thought, and I think you should close it.