keyboardio/Kaleidoscope #399: Qukeys: Add double-tap & hold for primary keycode repeat

algernon, 2018-10-18:
Originally by @gedankenexperimenter, as keyboardio/Kaleidoscope-Qukeys#29.

replicaJunction, 2020-04-02:
I know this is an old issue, but I hope it's not forgotten!

This is one of the only true losses in functionality I experienced after migrating an Ergodox from QMK to Kaleidoscope.

gedankenexperimenter, 2020-04-05:
I did implement this for Kaleidoglyph almost a year ago, but it was much trickier to do in Kaleidoscope at the time. It's probably easier now that Kaleidoscope's version of Qukeys has been rewritten, but I'm not sure. I hadn't forgotten about it, but I also hadn't heard anyone requesting the feature, either, so I had no motivation to spend time on it.

Note that this would not be identical to QMK's version (as I understand it); the quick press-release-press (and hold) would result in only a single keypress event sent to the host. Please correct me if I'm wrong, but I think QMK's version results in all three events being sent.

I now have a reason to work on the feature, but with my three kids needing constant attention at home now, I have very little time for it, I'm afraid.

gedankenexperimenter, 2020-10-01:
I just got another request for this on Discord. I will look into it soon.

gedankenexperimenter, 2020-10-01:
Here's how this will work:

1. When a Qukey is pressed, record the timestamp of when the key toggled on (we already do this).
2. If it resolves to its primary state, copy that timestamp before flushing the queue and sending the toggle-on report.
3. Delay the toggle-off event until either…
   a. the same key is pressed again (caution: layer changes?)
   b. a different key is pressed
   c. a timeout is reached (this timeout must be short enough that we won't trigger key repeat on the host)
4. If a quick double tap is detected, we need to delay again, because if it gets released quickly, we need to send two key press and release events to the host.

This is rather complex, but it does work, and it allows users on macOS to use the press-and-hold feature of Cocoa to access accented characters in the "normal" way without needing to subsequently delete the first letter.

gedankenexperimenter, 2020-10-03:
I've got a working version of this now. I'll clean it up and submit a PR when I get some more time. The code needs tidying and commentary, and the documentation needs updating, and it will be very useful to get some other people to test it out, but I think I've got it all working.

closed 2021-05-28 by algernon