keyboardio/Kaleidoscope #407: Leader: Leader needs to be released before it starts a sequence

algernon, 2018-10-18:
Originally by @Llamalland, as keyboardio/Kaleidoscope-Leader#9.

Llamalland, 2019-05-04:
Just jumping in on the two issues left that I have a stake in, since I haven't been in the community much lately. Still interested in sorting this out so that I don't have to wait for my leader key to go down when using it rapidly.

gedankenexperimenter, 2020-12-15:
@algernon — Do you recall why Leader sequences trigger on the keys toggling off, rather than on? I'm working on fixing this issue, and it's not obvious from scanning the code.

gedankenexperimenter, 2020-12-15:
@algernon — What do you think of the idea of having the whole Leader sequence sent to the output if no match is found in the dictionary, rather than just the non-matching key?

gedankenexperimenter, 2020-12-16:
I've got a fix for both issues described here (activating the Leader sequence on toggle-on rather than toggle-off, and the rollover issue at the end of a Leader sequence). However, it depends on #904. I'll submit it if and when that one gets merged. In the meantime, anyone for anyone who wants to try it out, it can be found here: https://github.com/gedankenexperimenter/Kaleidoscope/tree/b/Leader.rollover-from-end

gedankenexperimenter, 2021-05-26:
I think this is fixed by #1024. I want to verify it with a testcase to be certain.

gedankenexperimenter, 2021-05-27:
Confirmed that this is fixed by #1024.

closed 2021-05-28 by algernon