keyboardio/Kaleidoscope #415: OneShot: Repeat feature disappears after first layer oneshot cycle

algernon, 2018-10-18:
Originally by @BenjaminGalliot, as keyboardio/Kaleidoscope-OneShot#39.

algernon, 2018-10-20:
This is... getting complicated. It's easy to reproduce:

- Double-tap a one-shot layer key
- Observe that keys on the layer repeat
- Turn the sticky layer off
- Hold it (or double-tap it) again
- Observe that keys on the layer do not repeat anymore

I can only reproduce this after a sticky layer has been enabled once, not when just holding one long.

Funnily enough, all OneShot states appear to be the same in both cases.

algernon, 2018-10-20:
HAH. GOTCHA! Found the responsible code, now trying to figure out *why* normal, interrupting keys result in a masking. Probably a workaround for some rollover thing, which we may have fixed before.

algernon, 2018-10-20:
Ok, so this dates back to keyboardio/Kaleidoscope-OneShot#4, and this was a legitimate workaround at the time. Since then, we improved key lookup, and the case described in that report no longer apply: in case we hold the interrupting key, it will repeat the key that was on the oneshot layer, instead of repeating what's on the layer with the oneshot layer removed.

In other words, if we have `A` on the base layer, `B` on `TEST`, and we trigger `OSL(TEST)` (and do not hold the key), press the other key and hold it, it will repeat `B` instead of inputting `B` and repeating `A` as were the case in the aforementioned issue.

For this reason, I think we no longer need the masking at all, and it can be safely removed. Even in the `OSM` case, removing the masking does not cause any issues.

closed 2018-10-20 by obra

jamadagni, 2018-10-28:
Why is this marked “pending” if the PR has been pulled and fixes the bug?

algernon, 2018-10-28:
Because there's no automation that removes the tag, and it got the pending before being merged.

PR #434 (2018-10-20): OneShot: Drop the key masking workaround
Way back when, in order to fix the issue explained in keyboardio/Kaleidoscope-OneShot#4, we made it so that keys that interrupt a OneShot will be masked (unless certain conditions are met). Since then, we changed how keys are looked up, and the original problem is no longer present. This makes the masking unnecessary.

Begin unnecessary is good news, because that masking prevented key repeat for keys on a one-shot layer (even when the OSL key was held).

Fixes #415.
commit 4d42c2c6 (2018-10-20): OneShot: Drop the key masking workaround

Way back when, in order to fix the issue explained in
keyboardio/Kaleidoscope-OneShot#4, we made it so that keys that interrupt a
OneShot will be masked (unless certain conditions are met). Since then, we
changed how keys are looked up, and the original problem is no longer present.
This makes the masking unnecessary.

Begin unnecessary is good news, because that masking prevented key repeat for
keys on a one-shot layer (even when the OSL key was held).

Fixes #415.