keyboardio/Kaleidoscope #416: OneShot: Possible bug: OneShot.cancel() doesn't clear the Command key (OSM(LeftGui)) on Macs

algernon, 2018-10-18:
Originally by @joeynguyen, as keyboardio/Kaleidoscope-OneShot#44.

johngkhs, 2018-11-10:
Same issue here on Windows. OSM(RightGui) and OSM(LeftGui) both trigger a press of the Gui key after the oneshot timeout elapses.

Plugin order is:
KALEIDOSCOPE_INIT_PLUGINS(
  BootGreetingEffect,
  HostPowerManagement,
  LEDControl,
  LEDOff,
  NumPad,
  OneShot,
  EscapeOneShot,
  Macros,
  TopsyTurvy,
  ActiveModColorEffect
);

gedankenexperimenter, 2020-10-05:
I'm curious about the details of this one. Does anyone recall what the issue was, exactly?

obra, 2020-10-05:
This is the text of the original issue:

On Model01 1.23 firmware

I noticed that OneShot.cancel() doesn't clear the Command key (OSM(LeftGui)) on Macs. With my implementation, anytime I click any modifier key and then my new multi-modifier OneShot key, it clears out the previous modifiers and applies my new modifiers as "active". However, the exception is the Command key. It stays on and gets another "injection" of the Command key applied, causing it to be "sticky" while my Alt/Option key only becomes "active".

Let me know if that doesn't make sense and I'll try to re-explain.

```

void oneShotCommandAltMacro(uint8_t keyState) {
  /* disable OneShot sticky keys */
  if (
    keyToggledOn(keyState) &&
    OneShot.isActive() &&
    OneShot.isSticky(OSM(RightAlt)) &&
    OneShot.isSticky(OSM(LeftGui))
  ) {
    OneShot.cancel(true);

    return;
  }

  /* enable OneShot sticky keys */
  if (
    keyToggledOn(keyState) &&
    OneShot.isActive() &&
    OneShot.isModifierActive(Key_RightAlt) &&
    OneShot.isModifierActive(Key_LeftGui)
  ) {
    OneShot.cancel();

    OneShot.inject(OSM(LeftGui), keyState);
    OneShot.inject(OSM(LeftGui), keyState);

    OneShot.inject(OSM(RightAlt), keyState);
    OneShot.inject(OSM(RightAlt), keyState);

    return;
  }

  /* cancel other OneShot keys and enable these */
  if (
    keyToggledOn(keyState) &&
    OneShot.isActive()
  ) {
    OneShot.cancel();

    OneShot.inject(OSM(LeftGui), keyState);
    OneShot.inject(OSM(RightAlt), keyState);

    return;
  }

  /* enable OneShot keys */
  if (keyToggledOn(keyState) || !OneShot.isActive()) {
    OneShot.inject(OSM(LeftGui), keyState);
    OneShot.inject(OSM(RightAlt), keyState);
  }
}
This is happening with the implementation of my sketch at this commit - joeynguyen/dotfiles@610008f

```

gedankenexperimenter, 2020-10-05:
I'll look into this when I get some time.

gedankenexperimenter, 2020-10-05:
I've tried this out, and can't reproduce the problem with the current master branch. Both OneShot modifiers turn on and off simultaneously. However, I also found that this macro function doesn't have the desired effect (to make a OneShot combo of `Key_LeftGui` and `Key_RightAlt`). For me, the first tap makes both sticky, the second tap does nothing, and the third releases both.

I also tried the example OneShot macro, which is much simpler, but also doesn't have the desired effect (it works as a one-shot, but doesn't become sticky). I saw some other somewhat-irreproducible behaviour, effecting ordinary OneShot modifiers in the same sketch with these macros, too (a key seemed to become inoperative for a while, but I don't know what the sequence of events was that caused it.

I don't fully understand the behaviour I'm seeing, but I can say this: #905 provides a very simple and reliable way to get the desired effect, using `enableAutoLayers()`.

joeynguyen, 2020-10-06:
I was the one who initially filed the bug, but I've moved away from using OneShot in my workflow so I'm okay with closing this issue if you all agree.  The bug was initially filed many firmware versions ago so it may have been fixed now and wouldn't work with my previous implementation anymore.

gedankenexperimenter, 2020-10-06:
I think it's fair to say that this is not currently a problem. The code that originally revealed the problem still doesn't work, though my visual inspection makes me believe that it should, so a different bug is revealed. #905 does provide a solution to that secondary issue.

joeynguyen, 2020-10-07:
Sounds good, I will close.

closed 2020-10-07 by obra