keyboardio/Kaleidoscope #419: OneShot: OneShot should support more keys

algernon, 2018-10-18:
Originally by @MaxG87, as keyboardio/Kaleidoscope-OneShot#32.

dmikalova, 2018-12-09:
Another benefit to this would be enabling more layers - I'm trying out a layout that eliminates chording by using multiple layers and I've come up with 10 layers so far. Even as it grows I don't see myself going over 16 layers though.

algernon, 2018-12-09:
The problem with supporting more than 8 layers is that it would dramatically increase the code size. Right now, we're using a few 16-bit variables as bit maps, to store the state of one-shot keys. If we were to support more than 8 layers, we'd need to either use a 32-bit one (more memory, and much bigger compiled binary, because the platform is 8-bit, so 32-bit operations are done in software), or separate modifier and layer states and use a 16-bit vars for layers. The second increases code complexity significantly.

Before we explore such options, I think a big refactoring of OneShout would be required. I have a *feeling* that it is possible to come up with a design that doesn't suffer as many limitations as the current one. But the plugin needs to be heavily rebuilt for that.

gedankenexperimenter, 2020-10-04:
This would be addressed by #905, with `OneShot.enableAutoLayers()` (or `enableAutoOneShot()`), which lets OneShot treat `ShiftToLayer(N)` keys as OneShot keys automatically.

closed 2021-05-28 by algernon