keyboardio/Kaleidoscope #420: OneShot:  Using double OSM via macro makes them sticky

algernon, 2018-10-18:
Originally by @jamadagni, as keyboardio/Kaleidoscope-OneShot#38.

algernon, 2018-10-21:
Okay, I dug to the bottom of this, and the solution is both very simple, and a bit embarassing for me: You need to flip the order of `Macros` and `OneShot` in `KALEIDOSCOPE_INIT_PLUGINS`: `Macros` must come first. Otherwise `OneShot` will see the macro key before it gets processed, and since a macro key is neither a one-shot, nor a modifier, nor a layer key, it will be treated as an interrupt.

What goes on behind the secenes, if OneShot is first:

- The multi-OSM macro key is pressed
- OneShot ignores it, because there's no oneshot active
- Macros catches it, injects the one-shot keys
- The multi-OSM macro key is released
- OneShot sees that there's active oneshots, and this is a non-ignored key type, and immediately cancels the oneshots.
- Macros processes the release event, and injects the OneShot release event
- OneShot doesn't have any oneshots active anymore, but since it sees a oneshot key, cancels it anyway.

That's what happens with the current code. At the time this was reported, it may have worked differently.

With the recent OneShot changes (#435 in particular), canceling stickies should work too. With `Macros` before `OneShot`, the original example should work as well. I'll update the `OneShot.ino` example.

There's one remaining case left: double-tapping such a macro does not make the oneshots sticky. But that's something #414 is more suited to track with.

algernon, 2018-10-21:
@jamadagni Can you check if having `Macros` earlier than `OneShot` in the `KALEIDOSCOPE_INIT_PLUGINS` list helps?

algernon, 2019-01-25:
Cheers, thank you, marking this as resolved then.

closed 2019-01-25 by algernon

jalexvig, 2019-09-28:
@algernon Firstly, thanks for all your work on keyboardio and then making that work accessible to others. A lot of the keyboardio documentation is out of date (e.g. this, not using the `KEYMAP` macro, etc.) and requires a fair amount of time to debug (mostly stumbling around issues until I find one of your posts). It seems like the learning curve could really be sped up for new keyboardio users like myself by updating the documentation such that the examples work!

obra, 2019-09-28:
@Alex, If you see instances of out of date documentation, would you mind
opening bug reports about it?

> ᐧ

jalexvig, 2019-10-04:
@obra can do. The plugin code is now under this repo, but the old repos are the only place with the documentation (AFAIK) and those repos are archived... Where is the best place to raise issues/create PRs?

obra, 2019-10-04:
This repo is great. (Is
https://github.com/keyboardio/Kaleidoscope/blob/master/doc/plugin/OneShot.md
not
the doc you mean?)
ᐧ

jalexvig, 2019-10-04:
@obra perfect - I found the plugins from [the wiki](https://github.com/keyboardio/Kaleidoscope/wiki/Core-Plugins), so didn't realize that there was documentation in this repo. Updating the wiki to point to the docs in this repo would also be great. See #694