keyboardio/Kaleidoscope #423: OneShot: OneShot doesn't play nice with TapDance key

algernon, 2018-10-18:
Originally by @jhitch, as keyboardio/Kaleidoscope-OneShot#27.

gedankenexperimenter, 2020-10-05:
#639 was fixed. Does anyone recall what this issue was?

obra, 2020-10-05:
> I setup my period key to give me an exclamation mark when double tapped. This used the TapDance plugin. But the key combination [Key_Period] + [Space] + [OSM(LSHIFT)] + [Any letter] sometimes gives an unexpected result. This is the key combination that occurs at the end of every normal sentence. Sometimes, depending upon the speed of the typing, the result was ". .[Uncapitalized letter]"
> 
> To be sure that this was the issue, I disabled TapDance on my period key, and this behavior went away.

...

> 
> 1.) When I wrote the initial report, I had OneShot preceding TapDance in my order of plugins.
> I tried switching this order around. It does change the behavior of the keys, but the bug persists. If OneShot comes first, I can manage to apply my OSM to the TapDance key if the TapDance key was pressed first and held. This doesn't happen if TapDance is ahead of OneShot in my order of plugins. Not sure if this is another behavior that is unwanted, or maybe it is just better to keep TapDance first.
> 
> 2.) It is very hard for me to repeat this error on command. It's not just when I am typing fast, but there is something about the timing of these events (maybe related to the time the TapDance plugin is waiting to see if the key is pressed again? Sorry, but I'm not a programmer). I tried to do what you suggest, holding down my TapDance key, in this case a period, while I press the OSM key. It doesn't repeat the same result.
> My impression, though it is hard to say that I am 100% certain of this, is that I cannot possibly be holding down my period key when this happens. I think it is something else.
> 
> While it doesn't happen every single sentence, I estimate it happens about 1/3 sentences. I don't think it is user error. When I turn my period key back to a normal period, the OSM macro works every time as expected.
> 
>

gedankenexperimenter, 2020-10-05:
Thanks, @obra. I'll look into this one, as well.

gedankenexperimenter, 2020-10-05:
With OneShot registered first, I can confirm that the first bug described still exists, though I haven't found a very reliable way to reproduce it. I did observe some other odd behaviour, as well: sometimes, the TapDance key, which should produce a `.` has no obvious output. On one occasion, I saw seven periods rather than just one. I don't know yet if this is an interaction between the two plugins, or two (or more) independent bugs.

Then I tried holding down the TapDance key while pressing the OneShot key. The result was that, after a few attempts, I got a rapidly-toggling `.` key, even after releasing both keys. Karabiner-EventViewer event gets so far behind that it keeps printing them after I unplug the keyboard. It turns out that this is just a TapDance bug, however; merely holding the TapDance key alone causes it to get stuck in a toggle-on/toggle-off loop (probably because of the mid-cycle report that TapDance sends; I'll report that as a separate bug).

I also tried the second issue the original author reported (number 1 above), and with OneShot before TapDance in plugin order, the OneShot key fails to affect the output of the TapDance key (as I would expect). With TapDance first, things get more interesting: when I repeatedly press the OneShot, then the TapDance key (alternating), every third time I tapped the TapDance key, the OneShot shift didn't work. Once I paid attention better, it became clear that TapDance was consuming events such that OneShot never found out that the TapDance key was pressed, so stayed active until it timed out. And when the OneShot gets locked, I can see it appearing to toggle off every time I press the TapDance key, again because TapDance sends a mid-cycle report from its `onKeyswitchEvent()` handler.

My conclusion is that TapDance needs a rewrite. If I'm the one who does it, I'll use the `KeyAddrEventQueue` class I wrote for Qukeys, and eliminate the mid-cycle reports, and probably fix all of these bugs. I doubt I could make it backwards-compatible with people's existing configs, but I'm pretty sure I could make the new configurations simpler.

gedankenexperimenter, 2021-05-26:
I'm pretty sure that #1024 will fix this. I'll see if I can put together a testcase for it.

gedankenexperimenter, 2021-05-26:
I've experimented on this one some more. I think that the original form of this bug no longer exists, but I can reliably produce the current manifestation of it in the simulator with a testcase. In that testcase, tapping a TapDance key, followed by a OneShot key, followed by the TapDance key results in no output from the TapDance key, and can leave TapDance in an incorrect state for some time after that.

As expected, this bug is fixed by #1024, so I would close this issue when that PR is merged. However, in the process, I discovered a much smaller bug in the interaction with between the `KeyEvent` versions of TapDance and OneShot, where a pending TapDance key and a OneShot key are both resolved by the same event. In that case, the OneShot, which should apply only to the output of the TapDance key, applies to both that and the next keypress (the one that resolved the TapDance key).  Since that's not very clear, here's an example:

1. Tap `OSM(LeftShift)`
2. Tap `TD(0)` (which outputs symbols `x`, `y`, or `z`, depending on the number of taps)
3. Tap `Key_A`

The output should be `Xa`, but instead we get `XA`. This is fixable by having OneShot track more event state. Probably the best way to accomplish this would be to make it a subscriber to a global event queue. That would be a fairly big change, affecting several plugins, including TapDance, and might solve other plugin-interaction issues. It could also be fixed by having OneShot store more state on its own. Depending on the perceived severity of this individual bug, I could do the latter before eventually introducing the more general solution.

PR #1035 (2021-06-03): Add `afterReportingState()` event handler
This change adds a new `afterReportingState()` event handler function, which runs after HID report(s) are sent in response to a `KeyEvent`. This allows a plugin to act on an event after it's finished processing, rather than before. Right now, the only use case for it is fixing #423 (or what remains of it since #1024), in which there's a minor problem with an interaction between OneShot and TapDance.

The issue is that a OneShot key (`OSM(LeftShift)`) in its one-shot state, followed by a TapDance key sequence that terminates by being interrupted would r [... cut]
closed 2021-06-03 by algernon