keyboardio/Kaleidoscope #440: Redial: Support Redial during rollover

algernon, 2018-10-21:
This continues where #396 and #413 left off, in particular, we should support [this](https://github.com/keyboardio/Kaleidoscope/pull/438#issuecomment-431660603) scenario.

algernon, 2018-10-21:
I *think* what we need to do is track whether the previous key has been released yet when activating Redial, and if not, release it and perhaps mask the key before activating redial itself.

algernon, 2018-10-21:
Looks like we do support this behaviour as it is now, typing `stabbing` rapidly, with the second `b` Redial'd, it worked 10 out 10 times (at around 120WPM).

@LLamalland if you have the resources, testing current Kaleidoscope master to see if you still see the issue would be awesome (the previous PR was merged, so Redial should be improved overall).

algernon, 2018-10-22:
I tried my best to have overlap on the `b REDIAL` sequence.

Llamalland, 2018-10-22:
@algernon I don't know whether this is sufficient to say that I've tested this, but I updated everything I could on boards and libraries on arduino, got a vanilla firmware and put the redial code in. Same behaviour - redial works exactly as expected and desired when no other keys overlap with it, but trying to type "stabbing" led it to miss the redialled B when the initial B was still held down over the start of the redial keypress, and same with any other desired double letter.

Sorry that I'm not very familiar with how to do things on this - all the advice and instructions for more complex manipulation of arduino seem to be mac based. Maybe I should just uninstall Arduino from my system and reinstall it from scratch?

algernon, 2018-10-22:
Don't worry, knowing that the issue persists is a useful information for me, I can take it from here. I'll let you know if there's a possible fix to test!

As for upgrading things: what OS are you on? (Though, we should probably take this part to the forums)

Llamalland, 2018-10-22:
So the way I reproduce the behaviour is, pretty much as you described earlier, either: A) letter down, redial down, redial up, letter up or B) letter down, redial down, letter up, redial up. Assuming you don't do anything long enough for held repeats to trigger, you'll get only one of that letter and not the desired two.

While what is described doesn't happen every time I type something with redial, it happens maybe 1/3 of the time.

As for OS, I'm on Windows 10.

gedankenexperimenter, 2020-11-24:
I've written a testcase for this scenario in the simulator, and on rollover from another key to `Key_Redial`, Kaleidoscope outputs an empty report before the new report with the redialled keycode, as desired. The only reason I can think of why it might happen sometimes but not others scan order (maybe if `Key_Redial` was scanned first, it would work, but last and it wouldn't?). However, I explicitly tested that, and found no problem in either direction. Barring further reports of problems with Redial in rollover, I think this can be closed.

PR #987 (2020-11-24): Add testcases for Redial with rollover
I also simplified the Redial test suite sketch to make it easier to read.

Closes #440 (I'm pretty sure this demonstrates that rollover isn't a problem for Redial).
closed 2020-11-24 by obra