keyboardio/Kaleidoscope #457: USB-Quirks: idea: flash LED to represent protocol

tjkirch, 2018-11-08:
When switching between boot and report protocol, e.g. via MagicCombo, I'm not sure how to tell the current protocol, unless you're in a situation where only one can work like a BIOS.

It might be nice to flash the 'B' or 'R' LED to clarify the new protocol.

algernon, 2018-11-08:
I don't think this belongs to `USB-Quirks` itself, but I do agree it would be useful to show the state one way or the other. Luckily, this can be done by the part of the sketch that calls `USBQuirks.toggleKeyboardProtocol()`: you just need to look at `BootKeyboard.getProtocol()` to see which one's active after, and flash the appropriate LEDs.

Would updating the USBQuirks example to show how this can be done be enough?

obra, 2018-11-08:
I feel like maybe there’s a place for conditional action infrastructure in kaleidoscope. Something that would make this kind of thing easily doable as tiny plugins or one liners in the sketch.

tjkirch, 2018-11-08:
Both sound good to me :)  Thanks for considering it.