keyboardio/Kaleidoscope #481: Can't reflash firmware after upgrading to the latest Kaleidoscope

demizer, 2018-11-25:
Hi everyone!

I decided yesterday I wanted to give Chrysalis a try and so I made changes to my firmware at https://github.com/demizer/Model01-Firmware (a fork of algernon's firmware made a while back). I was able to get it to build and flash. Chrysalis is great, and I can't wait for it to become more fully featured. 

Unfortunately, after using Chrysalis, the "prog" button no longer works. I have nothing mapped to that button but it still does not work. This leads me to have to "hard re-flash" by holding down the prog key while plugging it in. The key does turn red, but when I press enter on another keyboard, I get:

```
$ make flash
BOARD_HARDWARE_PATH="/home/demizer/Arduino/hardware" /home/demizer/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash                                                                                                     
Building output/demizer/demizer (0.0.0-gv1.22-26-g10a1) ...
In file included from /home/demizer/Dropbox/src/arduino/keyboardio/Model01-Firmware/lib/Kaleidoscope-CapsLock/src/Kaleidoscope-CapsLock.h:4:0,                                                                                                                
                 from /home/demizer/Dropbox/src/arduino/keyboardio/Model01-Firmware/src/demizer.ino:39:
/home/demizer/Dropbox/src/arduino/keyboardio/Model01-Firmware/hardware/keyboardio/avr/libraries/Kaleidoscope/src/LEDUtils.h:19:2: warning: #warning The "LEDUtils.h" header is deprecated, the mode is included by default when using <Kaleidoscope-LEDControl.h>. It can be safely removed. [-Wcpp]
 #warning The "LEDUtils.h" header is deprecated, the mode is included by default when using <Kaleidoscope-LEDControl.h>. It can be safely removed.                                                                                                            
  ^
In file included from /home/demizer/Dropbox/src/arduino/keyboardio/Model01-Firmware/lib/Kaleidoscope-CapsLock/src/Kaleidoscope-CapsLock.h:4:0,                                                                                                                
                 from /home/demizer/Dropbox/src/arduino/keyboardio/Model01-Firmware/lib/Kaleidoscope-CapsLock/src/Kaleidoscope-CapsLock.cpp:1:                                                                                                                
/home/demizer/Dropbox/src/arduino/keyboardio/Model01-Firmware/hardware/keyboardio/avr/libraries/Kaleidoscope/src/LEDUtils.h:19:2: warning: #warning The "LEDUtils.h" header is deprecated, the mode is included by default when using <Kaleidoscope-LEDControl.h>. It can be safely removed. [-Wcpp]
 #warning The "LEDUtils.h" header is deprecated, the mode is included by default when using <Kaleidoscope-LEDControl.h>. It can be safely removed.                                                                                                            
  ^
- Size: firmware/demizer/demizer-0.0.0-gv1.22-26-g10a1.elf
  - Program:   26806 bytes (93.5% Full)
  - Data:       1482 bytes (57.9% Full)

To update your keyboard's firmware, hold down the 'Prog' key on your keyboard,
and then press 'Enter'.

When the 'Prog' key glows red, you can release it.

Looking for USB device with vid=1209 and pid=2301

Examining usb-Keyboardio_Model_01_kbio01-if00

  ID_MODEL_ID 2300 != 2301
ERROR: I couldn't find a USB device matching the keyboard's USB Vendor and Device IDs

I couldn't autodetect the keyboard's serial port.

If you see this message and your keyboard is connected to your computer,
it may mean that our serial port detection logic is buggy or incomplete.
In that case, please report this issue at:
         https://github.com/keyboardio/Kaleidoscope
make: *** [/home/demizer/Arduino/hardware/keyboardio/avr/build-tools/makefiles//rules.mk:80: flash] Error 1
```

Rebooting does not resolve the problem.

My copy of https://github.com/keyboardio/Kaleidoscope-Bundle-Keyboardio is latest:

```
$ git show --summary
commit d5aaf49427de0acc26a1acba11028bb6ccb4c520 (HEAD -> master, origin/master, origin/HEAD)
Author: Gergely Nagy <algernon@keyboard.io>
Date:   Tue Nov 20 08:01:40 2018 +0100

    Update Kaleidoscope to latest master
    
    Signed-off-by: Gergely Nagy <algernon@keyboard.io>
```

I checked the kaleidoscope version is latest:

```
$ git show --summary
commit 819ab0760f36ded1ac8b871a40a007e2bc049332 (HEAD -> master, origin/master, origin/HEAD)
Merge: 3c16dc05 ef94658b
Author: Jesse Vincent <jesse@fsck.com>
Date:   Fri Nov 23 09:28:32 2018 -0800

    Merge pull request #477 from keyboardio/focus/new-api
    
    FocusSerial: New, simplified API
```

and my udev rules are installed:

```
$ cat /etc/udev/rules.d/99-kaleidoscope.rules
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1209", ATTRS{idProduct}=="2300", SYMLINK+="model01", ENV{ID_MM_DEVICE_IGNORE}:="1", ENV{ID_MM_CANDIDATE}:="0"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1209", ATTRS{idProduct}=="2301", SYMLINK+="model01", ENV{ID_MM_DEVICE_IGNORE}:="1", ENV{ID_MM_CANDIDATE}:="0"
```

Any help would be greatly appreciated. Thanks!

I am using Arch Linux.

algernon, 2018-11-26:
When trying to flash normally, are you holding `Prog` long enough? I found it usually needs to be held a bit longer than when it starts to glow red, and usually hold it until the flashing animation starts.

It might have worked before because the detection was different and a tiny bit faster, I suppose.

If holding longer still doesn't work, then we're looking at some funky bug indeed.

demizer, 2018-11-26:
@algernon Thanks for the reply

I tried holding the prog key for a minute and it never turned red. I confirmed the key is set to "transparent" for all layers in Chrysalis.

algernon, 2018-11-26:
That's weird, 'cos Chrysalis doesn't do anything that would result in something like this. I'll see if I can find what's going on. I might end up asking a few more questions from you, or things to try that might reveal more information, if you don't mind.

demizer, 2018-11-26:
No worries, I would be glad to help!

algernon, 2019-01-20:
While running `make flash`, do you have any other tool open that might open the serial port on the keyboard at the same time (Chrysalis, or `tools/model01-commander.py` or the like)? Those can - and do - interfere with flashing.

(Apologies for being slow to respond, I was busy with Chrysalis stuff)

demizer, 2019-02-13:
Oh I feel incredibly dumb. I was doing it wrong. I was holding down the prog key expecting it to turn red, then I would press the enter key. After a little RTFM, I held down the prog key, then pressed enter, and then it turned red and the firmware flashed.

closed 2019-02-13 by demizer

demizer, 2019-06-03:
This happened to me again, and despite holding the prog key and then pressing enter, the prog key never glowed red and then I would get the USB Device ID error. To get it programming working again, I had to unplug the device, hold the prog key, then plug it back in. The prog key glowed red, but I still could not flash the device until the prog key stopped glowing. After that though, I was able to flash again.