keyboardio/Kaleidoscope #482: Consider adding warnings to documentation for users using latest stable version in Arduino IDE

hovissimo, 2018-11-26:
I spent a long time tonight trying to figure out why my Model01 was exhibiting the fixed bug https://github.com/keyboardio/Kaleidoscope-TopsyTurvy/issues/11 (marked as dupe of #5 and #6 in that repo). The recent switch to the monorepo aggravated my confusion, but in a nutshell the "Getting Started" info I found in the wiki which leads me to using v 1.23 installed via the Arduino IDE gives me the old buggy version of TopsyTurvy.  Worse, I was extremely confused about how to get the latest TopsyTurvy and initialize it because the method of initializing plugins between current master and 1.23 has changed.

Eventually I gave up on the Arduino IDE on Windows and followed the instructions for a CLI build on Linux (and only at this point realized that KALEIDOSCOPE_INIT_PLUGINS is the new and correct technique). Now I'm up and running just fine, but it wasn't a great experience for my first time trying to customize my keyboard.

Unless current master is released as an IDE build fairly soon, adding some warnings to the documentation would be a boon to confused noobies, especially those of us unfamiliar with c/c++ and Arduino under the hood.

Specifically, here's sort of a story format of how my evening has gone.
1. https://github.com/keyboardio/Kaleidoscope/wiki/Install-Arduino-support-on-Windows
2. https://github.com/keyboardio/Kaleidoscope/wiki/Edit-a-keymap
3. Google:  how to invert number row/symbols
4. Found reference to TopsyTurvy
5. Used TopsyTurvy example in Arduino IDE
6. Experience the bug now fixed in current master
7. Google: "keyboardio TopsyTurvy repeat"
8. Find https://github.com/keyboardio/Kaleidoscope-TopsyTurvy, go check the issues to see if it's been reported/solved
9. Find bug 11, follow the dupe link, see that it's fixed in that repo
10. Suspect that the version I have is out of date
11-16. Super confused about how to see/verify libraries installed by Arduino IDE on Windows
17. Start re-reporting that bug in this repo, and realize that I should try the CLI build first to be sure
18. Get it working after following Linux instructions starting at https://github.com/keyboardio/Kaleidoscope

algernon, 2018-11-26:
> Unless current master is released as an IDE build fairly soon, adding some warnings to the documentation would be a boon to confused noobies, especially those of us unfamiliar with c/c++ and Arduino under the hood.

FWIW, not current master, but a reasonably recent snapshot has been released as 1.93-beta. Nevertheless, we should update the Wiki to suggest installing that, or git master, indeed.

obra, 2020-06-08:
We've pushed updates since then. Closing out

closed 2020-06-08 by obra