keyboardio/Kaleidoscope #484: Stop SpaceCadet when multiple keys are pressed

gglanzani, 2018-11-26:
Hi,

I'm using SpaceCadet to input `({[]})` (bound to ctrl, etc.).

However, even when the timeout it low, I find myself hitting things such as ⇧ ⌘ L (a random shortcut), and having it not fire because the keyboard thinks I want to use SpaceCadet.

Could we "disable" the waiting when another key is pressed together with one of the Cadet keys?

I.e. if I hit ⌘ (space cadet key) and ⇧ (non space cadet key), it should immediately timeout SpaceCadet. Is something like this possible?

Cheers

gedankenexperimenter, 2018-12-17:
You could use Qukeys instead of SpaceCadet to get the functionality you want without this problem.

gedankenexperimenter, 2018-12-18:
With Qukeys, what matters is the order in which the keys are *released*. If you're someone who releases the two keys simultaneously, there's a race condition. However, you can use `Qukeys.setReleaseDelay()` to set a grace period (~20ms is probably good), it is likely to work more reliably for you.

gglanzani, 2018-12-19:
@gedankenexperimenter I have that as well, I've tried increasing it to 30, to see if it reduces the chances to mistype. Thanks!

gedankenexperimenter, 2019-05-06:
I have now rewritten Qukeys (#640), and one of the changes I made was SpaceCadet emulation. In that branch, a Qukey object defined on a keymap entry that's a modifier (or layer shift) will use the primary key value when held, and the alternate value when tapped. It will also use the primary (modifier) value if there's any rollover to another key, so it should solve this problem.

gedankenexperimenter, 2021-05-26:
I believe this is fixed by #1024. I want to verify it with a testcase first.

gedankenexperimenter, 2021-05-27:
Confirmed that this is fixed by #1024.

closed 2021-05-28 by algernon