keyboardio/Kaleidoscope #488: Make it possible to use Fn to toggle to different layers when held vs when tapped

obra, 2018-12-05:
@algernon Do we have a space cadet shift style way to make Function shift to different layers depending on whether it's tapped or held?
algernonToday at 4:08 PM
so it would have two actions: toggle to layer A when tapped, shift to B when held?
obraToday at 4:08 PM
yes.
algernonToday at 4:08 PM
I don't think we have that (yet)
obraToday at 4:08 PM
I'm trying to implement half-qwerty for a customer who's temporarily disabled
algernonToday at 4:08 PM
if you open an issue, I'll PR it tomorrow morning.
obraToday at 4:08 PM
ok

algernon, 2018-12-05:
On second thought, SpaceCadet can already do this: just set up a mapping like `{LockLayer(A), ShiftToLayer(B), 500}`, and have `LockLayer(A)` on the keymap + SpaceCadet enabled.

(Not tested, but should work according to the docs)

gedankenexperimenter, 2018-12-19:
Incidentally, Qukeys can also do this. It can even do two different `ShiftToLayer()` values on the same key, though the utility of this is dubious, as the "tap" version would only be useful when rolling over, and would probably require a fairly long timeout to be used reliably.