keyboardio/Kaleidoscope #498: We need a Kaleidoscope plugin that flashes the avr keyscanner firmwares when triggered.

obra, 2018-12-11:
Right now, updating the avr keyscanner firmware requires flashing a custom tool onto the Model 01.  That tool is small enough to become a Kaleidoscope plugin we could ship as part of a regular firmware update, but it needs extensive rewriting to work. The standalone tool is attiny_flasher.ino in the built output of the avr_keyscanner repo. As of 9bc65bdd5c010879734b00f43f6edeadefe4f761, master of that repo is the correct rev for us to be porting from.

algernon, 2018-12-13:
What should be the behaviour here? The plugin would provide a `Model01ScannerUpdate` object, with a single method: `Model01ScannerUpdate.update()`. Or perhaps we can move it to the hardware plugin (if it can be optimized out if unused), in which case it'd end up being `Model01.updateScannerFirmware()` or somesuch.

My question is mostly about what is the expected outcome of this function?

Should it reboot the keyboard back to the regular firmware when finished? (I'd say yes)

Is there a way to signal progress with LEDs while flashing is ongoing? (probably not?)

What shall happen upon failiure?

algernon, 2018-12-13:
Oh, and for the record: we're talking about a plugin that flashes a particular version of `avr_keyscanner`. To flash a different version, the plugin will need to be updated. I think this task should involve writing a tool that helps us update the plugin, too.

algernon, 2019-01-20:
I have closed #500, which attempted to implement this, because I think we can do better than embedding the ATTiny firmware within one's normal sketch.

My counter-proposal to including the firmware updater in one's sketch is this:

- Implement a `firmware.dump` Focus command, which dumps the firmware, from zero 'till whatever the MCU supports, the whole thing.
- Update the ATTiny updater firmware to output machine-friendly progress
- Teach Chrysalis to dump firmware (`firmware.dump`), ship an ATTiny updater firmware with it, and then we can ask it to dump the current firmware, flash the attiny updater, wait for it to finish, then flash the old one back.

This setup is also easily scriptable, if one doesn't want to use Chrysalis. We can cook up a small shell script that does essentially the same thing, without having to install 100s of megabytes of JavaScript for Chrysalis.

This assumes that we can both dump the firmware, and flash it back. Even if we can't, I'd rather not embed the updater into one's sketch. It's big, and used very, very, very, very rarely. We should make it easier to do the attiny update rain-dance without having to ship it with one's own sketch isntead.

algernon, 2020-01-06:
Well, I think this can be closed by now. First, we have the [`FirmwareDump`](https://github.com/keyboardio/Kaleidoscope/blob/master/docs/plugins/FirmwareDump.md) plugin, and the [`kaleidoscope::util::flasher`](https://github.com/keyboardio/Kaleidoscope/blob/master/src/kaleidoscope/util/flasher/KeyboardioI2CBootloader.h) tools which cover both the firmware dump idea above, and the in-sketch updater.

The flasher currently does not support AVR, and I doubt it would be very useful there due to space constraints, but the foundations are there.

obra, 2020-01-07:
I'd really like it to support AVR. We've got a lot of Model 01s in the
field that could really use the new keyscanner firmware.
ᐧ

algernon, 2020-01-07:
I can do it, but space constraints will make it very hard to create a sketch with it, I think. But I guess the best way to see that is doing it!

commit 66f83151 (2018-12-14): Model01: Implement an (optional) scanner update mechanism

The new functionality updates the ATTinys, with a hard-coded firmware. The
currently hardcoded firmware is noted in the sources.

The tools to update the updater are not included with this patch yet, this is
merely a hand-ported solution.

Fixes #498.