keyboardio/Kaleidoscope #516: HostPowerManagement: `toggleLedsOnSuspendResume` doesn't work

jamadagni, 2018-12-26:
Hello. Currently using fbe335e1ffd00ac0d587a23300cc7a891e2bc2ab. Using `toggleLedsOnSuspendResume` from [vanilla sketch](https://github.com/keyboardio/Model01-Firmware/blob/master/Model01-Firmware.ino) as part of [mine](https://gist.github.com/jamadagni/afe3848e1e5f95343e8e84fb6c60f4e7). 

The toggling doesn't work. That is, even if my computer goes to sleep, the LEDs don't go off. No change is seen on resume either.

Before the change to the monorepo it was working fine. That's all I know.

Please check and fix. Thanks.

tiltowait, 2019-01-07:
I have the same issue—my LEDs won't turn off when my system (macOS 10.14) sleeps.

> However during host sleep the keyboard becomes unresponsive in the sense that LED mode changing or other keyboard-only operations don't work until the host is awoken.

I do not have this issue, however.

algernon, 2019-01-20:
> However during host sleep the keyboard becomes unresponsive in the sense that LED mode changing or other keyboard-only operations don't work until the host is awoken.

Does this mean you are also unable to wake the host with the keyboard, and have to use another device?

algernon, 2019-01-20:
Ok, I could reproduce the issue with LEDs not going off, it is caused by cac8e0f6149f17f9d9f40ba4bc85939706a9a6a4. The fix for that is to change your sketch to pause *after* `LEDControl.syncLeds()`. See keyboardio/Model01-Firmware#76.

LED mode changing not working while suspended is intentional, as we disable all kinds of LED updates during suspend, so that part is not a bug. If you want to turn LEDs off, but still allow LED mode changing, you can remove the `LEDControl.paused = true` line. Mind you, key presses will wake up the host, so you won't be able to play with LEDs while the host sleeps.

jamadagni, 2019-01-24:
Moving `LEDControl.paused = true` fixed the problem. Thanks!

closed 2019-01-24 by jamadagni