keyboardio/Kaleidoscope #518: A makefile target to build an 'amalgamation' of a sketch

obra, 2018-12-29:
To use compiler-explorer for optimization, I'd really like to have a makefile target that bundles everything above the avr headers into a single file that I can load into the explorer

algernon, 2018-12-29:
It's not easy to discover which files get used during a build, but there are ways that might work. I wouldn't try to replicate what Arduino is doing, because that's going to be brittle. So the next best thing is to run a verbose build, pick out the source files used for compilation, pick all the headers out from them, and merge. We need to replace `#pragma once` stuff though, and finding the headers will not be without issues either.

I'll see what I can do.

obra, 2018-12-29:
Rather than replacing the #pragma once stuff, could we let the preprocessor
do it?

algernon, 2018-12-29:
Hm.... we could just merge all the c/cpp files, and let the preprocessor do the includes and deal with pragmas, indeed.