keyboardio/Kaleidoscope #534: Move ::isModifierKey() to hid::isModifierKey()

tiltowait, 2019-01-20:
`hid.cpp` contains a function:

```c++
// isModifierKey takes a Key and returns true if the key is a
// keyboard key corresponding to a modifier like Control, Alt or Shift
// TODO: This function should be lifted to the Kaleidoscope core, somewhere.

bool isModifierKey(Key key) {
  // If it's not a keyboard key, return false
  if (key.flags & (SYNTHETIC | RESERVED)) return false;

  return (key.keyCode >= HID_KEYBOARD_FIRST_MODIFIER &&
          key.keyCode <= HID_KEYBOARD_LAST_MODIFIER);
}
```

This is a useful potential API call that a plugin might want to use. Given the `TODO`, it isn't clear to me why `isModifierKey()` couldn't just be part of the `hid` namespace and moved outside of the anonymous namespace in which it currently resides. It would fit in well with `isModifierKeyActive()` and `wasModifierKeyActive()`, which *are* part of the `hid` namespace.

Unless there's a reason it can't/shouldn't be moved there, I can go ahead and create a PR.

gedankenexperimenter, 2019-01-27:
I think I'm the one who wrote that function, and put it in that anonymous namespace. I believe that I put it there because it was a part of the fix for a rollover bug (or two?), and most of the code that came with it seemed best to leave not exposed as part of the API. I agree that this function doesn't fall into that category, and it would be appropriate to add it to the API so that plugins could access it.

obra, 2020-06-08:
This moved with a refactor sometime in the past year.

Resolving.

closed 2020-06-08 by obra