keyboardio/Kaleidoscope #552: Syncing up with Dygma

algernon, 2019-01-25:
In the not too distant future, we'd like to sync up with @Dygmalab's forks. I've had a chat with @mattvenn the other day, and went through most of the repos they forked or made, to figure out what needs to be done on either side to be able to sync up. We identified a couple of pain points. I'm listing them here instead of opening issues against Dygma repos, because this makes for a nicer overview, and some of the work will have to be done on the Kaleidoscope end.

- [ ] The Arduino Core Dygma uses does not have `Serial`, but `USBSerial`, with similar, but not quite the same API. A bunch of plugins were forked that used `Serial` directly. We no longer do this, all such stuff goes through `Focus` (except for `Steno`, which has to use Serial directly, but we're limited to two plugins which is a big improvement already). I think the best way forward is to implement an AVR-compatible `Serial` API in either the ARM Arduino Core, or the Raise hardware library. It should be a pretty thin one. Once done, any plugin that uses Focus or Serial will immediately work with the Raise.
- [ ] EEPROM on the ARM MCU of the raise is emulated, and has a different API than AVR's `EEPROM` object. There is a wrapper, but we still need additional code here and there. Since we're moving towards abstracting away EEPROM access (#344), this will be less of an issue. We'll still need to write a bit of code for Raise to bind the EEPROM emulation to the new PersistentStorage API, but like with Serial, once this is in place, there will be no need to fork individual plugins.
- [ ] LED Underglow. The Raise has both per-key LEDs and Underglow. Our Underglow API is still in development (#551), and we need to make sure the proposed API is suitable for Raise too.
- [ ] The forked KeyboardioHID could be revisited. It was forked because of BootKeyboard issues, which we should have fixed since. There are some changes that were made because the ARM core lacked certain functionality (like `USBDevice.wakeupHost()`). I think those should be added to the Core, even if they're no-op for the time being. Similarly, there are subtle differences between the AVR and SAMD cores: the former has `PluggableUSB.h`, the latter `USB/PluggableUSB.h`. Like in the previous case, I think this is best corrected in the SAMD core, by adding a `PluggableUSB.h` header there (which includes `USB/PluggableUSB.h`, to not diverge much from SAMD upstream), or by adding an `-IUSB` or similar to Raise's default CFLAGS.

Thankfully, most of this is reasonably easy to sync up, the major pain points are EEPROM and Underglow, both of which have work under progress, and we'll make sure they suit the Raise's needs too.

obra, 2019-01-25:
In cases where the SAMD arduino core and the AVR arduino core differ, I strongly suspect that the SAMD way is the 'new arduino' way. 

I suspect that we're not going to have a ton of luck getting Arduino to add these changes to their cores.

Consequently, I expect that processor-based #ifdefs are going to end up being a much easier path forward.

I believe that Arduino's core engineers have contributed SAMD Arduino support to Nico Hood's USB HID implementation. We can probably run with what they're doing there.

algernon, 2019-11-15:
I'm closing this one, because except for Underglow, we're pretty much synced up, and Underglow is sort-of supported by now. Not in a nice way yet with proper APIs, but the Raise can make use of it regardless.

closed 2019-11-15 by algernon