keyboardio/Kaleidoscope #556: Keyboard events come out of a mixed keyboard+pointer HID interface

dakkar, 2019-02-01:
Device: KeyboardIO Model01, checked out Kaleidoscope bundle at commit keyboardio/Kaleidoscope-Bundle-Keyboardio@15cc47a56aaeed13dcb59137f5a6755ba97bbb08

Host: three different Linux machines, running Gentoo, kernel 4.14 or thereabout, X11 server xorg version 1.20.3, input driver xf86-input-evdev 2.10.6

The keyboard shows up as 3 different HID interfaces, enumerated in this order:
* keyboard
* keyboard + relative pointer
* absolute pointer

`lsusb` shows all three, the X server recognises them, the evdev driver sets  them up, the evdev-us-basic XKB keymap is applied to both keyboard devices, and I can type.

But: the effective keymap is not evdev-us-basic, it's the evdev-us-dvorak of the *other* USB keyboard I have plugged in.

Disabling the xinput devices one by one reveals that keyboard events are coming from the keyboard+relative device, which shows up in the xinput tree as a pointer device, and this seems to confuse either the xinput or the XKB layer.

If I toggle the Model01 to boot protocol, events come out of the keyboard-only device, and the correct keymap is used.

I have re-flashed with [all the mouse features disabled](https://www.thenautilus.net/cgit/keyboardio-model01/commit/?id=881a8aaf74b5e4d8337047c4d26454bb08726ead), and I still get a keyboard+pointer HID device, and the same problem. Even [disabling boot protocol support](https://www.thenautilus.net/cgit/keyboardio-model01/commit/?id=59ad017752cd8674c34540709a6227a583be12ca) doesn't change that.

My summary of the situation (which may well be wrong):
* there's some bug between xinput and XKB, that prevents keymaps applied to pointer+keyboard devices from working
* the evdev input driver might be confused by the mixed device
* Kaleidoscope should probably not set up such a mixed device in the first place

I'll reconfigure the X server to use the libinput driver instead of evdev, to see if it helps, and report back.

dakkar, 2019-02-01:
I just checked with `xf86-input-libinput`: that driver splits the single mixed devices into a pointer and a keyboard, and everything works.
If I manage to re-implement the hacks I needed for the *other* keyboard, I can live with this.
I'd still like to know where, in Kaleidoscope, the mixed device is set up, and why it is that way.

algernon, 2019-02-01:
> I'd still like to know where, in Kaleidoscope, the mixed device is set up, and why it is that way.

It's in [KeyboardioHID](https://github.com/keyboardio/KeyboardioHID). I guess we need to split Mouse off of MultiReport, like we did for AbsoluteMouse. Or at least make that an option (and perhaps the default).