keyboardio/Kaleidoscope #560: Weird issue after PrefixLayer plugin moved to v2 plugin API

ilianaw, 2019-02-05:
Howdy! I don't know if this is a bug in Kaleidoscope or [the plugin I use, PrefixLayer](https://github.com/jamesnvc/Kaleidoscope-PrefixLayer).

algernon, 2019-02-05:
I think this isn't the v2 API, but the way we treat `LSHIFT(foo)` (we recentishly made changes around that area). I'll have a look later today.

gedankenexperimenter, 2019-02-11:
@algernon — Do you think this is a result of the rollover bug fix(es)? If I have time, I might look into it this week.

algernon, 2019-02-11:
That's my suspicion, yes.

gedankenexperimenter, 2019-02-12:
@algernon, I believe you are correct, based on [this comment](https://github.com/keyboardio/Kaleidoscope-HIDAdaptor-KeyboardioHID/pull/5#discussion_r197097670) I made in the mod-flag-rollover fix. I will need to go over that code again to sort out what the problem is, however, since it is no longer fresh in my mind. This could also be the source of a problem reported with SpaceCadet on Discord, but I'm not so sure about that guess.

gedankenexperimenter, 2019-02-12:
I woke up this morning 99% sure I know how this is happening, and 90% sure it can be fixed with changes to PrefixLayer.

There is another, deeper problem that exists with PrefixLayer (and any other plugin that sends reports mid-cycle) but not Qukeys (the only plugin I know of that works around that problem). But the converse is also true (Qukeys only works with MultiReport, and bypasses the HIDAdapter in order to work around the problem).

I think the only way to fix *both* issues is to add functionality to either the HID or HIDAdapter module so that, when a mid-cycle report is being generated, a backup copy of the current keyboard report is saved in some form, and any additional `Key`s are added to a copy of the *previous* report instead of the incomplete current report in order to not send an incomplete report, which results in unintended repeat keypress events in the output.

gedankenexperimenter, 2019-02-12:
The most obvious way to implement it would be for each HID keyboard object (in the case of KeyboardioHID, `Keyboard` and `BootKeyboard`) to store a third copy of the report and have methods for "stashing" (to borrow `git` terminology) the current report and replacing it with a copy of the previous report and for restoring the stashed, incomplete report. Plugins could then call `hid::stashReport()`, then add or remove keycodes (e.g. calling `handleKeyswitchEvent()`), send the mid-cycle report, and finally call `hid::restoreReport()` before proceeding.

That `hid::stashReport()` function is probably the best place to store the stashed report (rather than in the HID module itself), in an anonymous namespace of the HIDAdapter module. That way, it can also fix this bug by backing up the other state variables used to fix the rollover bugs.

algernon, 2019-02-12:
What happens when we want to hold the mid-cycle report? What if we don't know we want to hold it until the next cycle or two?

I'm thinking of the case of having a - say - TapDance key that's `a` on first tap, `Shift` on second. To get `B`, you'd `TD TD+hold B`... hmm... in case of tapdance, if being interrupted, we can look at whether the tapdance key is still held, and not restore the report if it is. When timing out, likewise.

This could work. Might even make some plugins (TapDance and OneShot come to mind) simpler, but that's a hunch.

gedankenexperimenter, 2019-02-12:
In the TapDance example, isn't it the key *event* that you want to delay, not the HID report? I may be misunderstanding what you mean by "hold"…

gedankenexperimenter, 2019-02-12:
So, it turns out that my guess about this problem was incorrect. It *is* a problem related to the changes made to fix the rollover bugs with modifier flags, but what was really happening was that there was a spurious `return` statement in PrefixLayer that was preventing the initial keypress event from getting through, thereby allowing the mod flags bitmask to get set properly.

gedankenexperimenter, 2019-02-12:
See jamesnvc/Kaleidoscope-PrefixLayer#3 for the PR that fixes this problem.

@algernon – I still think there should be some changes made to Kaleidoscope, _et al_ to enable safer sending of mid-cycle keyboard reports, but I think this issue is safe to close. I've submitted both an issue and a PR against the PrefixLayer repository to track and fix this bug.

obra, 2019-02-12:
@gedankenexperimenter If you haven't already done so, will you open a ticket against kaleidoscope describing the use cases for being able to send a mid-cycle key report?

My first gut reaction is that rather than adding a stash into what was intended as a very thin layer to send key reports over USB, we should be exposing more of the internal data structures so that code that really wants to monkey with USB reports has enough rope to do so.

closed 2019-02-12 by obra

gedankenexperimenter, 2019-02-12:
I will do so, probably late this evening. I've got a set of PRs almost ready, too, so you should be able to look at the code that I'm proposing shortly thereafter.