keyboardio/Kaleidoscope #564: Add a way to move/zap to a layer

algernon, 2019-02-06:
We can shift to or lock a layer, both of those keep all other layers active. We should be able to move to a layer (activating it, and deactivating all others) too, without using macros. We already have `Layer.move`, just need to expose it as a Key.

(Thanks @samaaron for the reminder!)

algernon, 2019-12-31:
This is something Dygma is interested in too, pull request coming soon.

algernon, 2019-12-31:
I experimented with adding `MoveToLayer(n)`, which does exactly what it says: moves to that layer, and disables everything else. The problem with this approach is that we'd also need to make something similar to `ShiftToLayer`, which moves only momentarily. I'm struggling to even come up with a reasonable name for that.

Struggling with naming, and having two _very_ similar things is a sign that we might not be using the right approach. I mean, the difference between the two is that one disables all other layers, and the other does not. I don't really see a scenario where one would want to use both approaches within the same keymap, so... why don't we have a knob in `Layer` instead? Lets call it `Layer.singleLayersOnly(bool enabled)`. If enabled, when switching layers, it would disable all others.

Now, we do have `Layer.activate()` and such, and I wouldn't want to change the behaviour of those, based on a setting. That wouldn't make much sense. So we'd introduce `Layer.moveTo()`, or `Layer.switchTo()`, or something along those lines. Depending on whether `singleLayersOnly` is enabled or not, these would either `.move()` or `.activate()` the given layer. In other words, we'd add a new method on top of the existing ones, and the setting would _only_ affect this.

This way we remain 100% backwards compatible. Any code that uses the currently present APIs would work the same way, no matter the setting. New code could opt to use the new methods.

The big question in this case, is what to do with `LockLayer` and `ShiftToLayer`? If we want to remain 100% backwards compatible, even when the new setting is enabled, we shouldn't change those. But if we don't, we need to introduce `MoveToLayer` and the `ShiftToLayer` counterpart, and we're back to square one with having to name it. As such, if we take this approach, I'd change `LockLayer` and `ShiftToLayer`. If no code changes the knob, then things will work as they do today. If one adds a few lines to change the setting, then they'll behave differently, but that needs conscious change from the end user.

algernon, 2020-01-01:
Ok, there are other issues with the knob approach, namely `ShiftToLayer()` behaviour. But that's kind of a problem with every other approach as well: it doesn't make sense. `ShiftToLayer()` only makes sense if we have multiple layers active, because if we don't, where do we switch back to? Do we keep track of the last active one? What if we press `ShiftToLayer(1)` first, hold it, then hold `SwitchToLayer(2)`, and release either. Where do we end up then? Granted, we can make it so that `ShiftToLayer()` enforces the layer while held, not just when the key toggles on, so if we release `ShiftToLayer(2)`, we'll end up on layer 1, since `ShiftToLayer(1)` is still held. But what if we release that too? We'd need to keep track of the previous non-shifted layer. And I'm pretty sure there would be other complicated corner cases.

With this in mind, I don't think making a variant of `ShiftToLayer()` that moves to the layer instead of activating it on top of the others makes sense. And if we don't have to deal with `ShiftToLayer()`, then my original approach in #779 is fine.

Coincidentally, the code is simpler that way than it would be with the knob, if only `LockLayer`/`MoveToLayer` is affected.

obra, 2020-01-01:
I think it would be valuable right now to start writing out test cases in pseudo code.

obra, 2020-01-02:
At a user experience level, it seems like the only difference between MoveToLayer and LockLayer is transparency. Is that right?

algernon, 2020-01-03:
Yes, that's pretty much the only difference.

obra, 2020-01-03:
I wonder if it would make more sense to reconsider transparency or to make transparency an option for shifting and locking.  getting rid of transparency could simplify a whole bunch of code.

algernon, 2020-01-03:
> I wonder if it would make more sense to reconsider transparency or to make transparency an option for shifting and locking. getting rid of transparency could simplify a whole bunch of code.

Hm. Hmm. If we keep the current behaviour of activating layers and keeping others active, and just disable transparency, that'd be an option. However, there's one case where that's _very_ different than `MoveToLayer`: when going from a higher layer to a lower. With multiple layers active, and transparency disabled, you cannot go lower, without disabling the higher ones. With moving layers, when there are _no_ other layers active, you can.

One of the things people find very surprising with the current layer behaviour (which is shared by QMK, where people are often similarly confused) is that you can't go to a lower layer. You can activate one, but higher layers would shadow parts of it. And this becomes incredibly mind boggling very fast, until you "get" layers.

Disabling transparency, but keeping layers active would not solve this. We wouldn't be much further ahead than we are now.

On the other hand, if we move layers, then `ShiftToLayer` becomes an issue. We can't easily "momentarily" move to another layer, because then we'd need to keep a stack of layers to go back to, and that's both complicated and fragile. Why we'd need a stack? Imagine we're keeping only the previous layer index around. Lets start on L#0, and shift to L#1 (`prev_layer_ = 0`), then shift to L#1 from there, while still holding L#1 (`prev_layer_ = 1`). Then release L#1... oops, what should `prev_layer_` be? But if we release L#2 first, and then L#1 - same problem. If we prevent shifting to another layer while one shift is active, that would solve this problem, but would cripple layer shifts, in my opinion. This is why I haven't implemented a shifting counterpart to `MoveToLayer` in #779.

algernon, 2020-01-03:
I just had a terrific idea.

As far as I see, we have to major ways of doing layers: one is the stacking them in index order with transparency (what we do now), and the other is stacking them in activation order without transparency (which is less powerful, but perhaps more in line with what people new to layers would expect).

The big downside of the second is that it requires more memory. For the first, we need a single uint32_t (4 bytes) to store all layer state. For the second, we need 32 uint8_t (32 bytes), a 28-byte RAM increase, and likely a bit more code for bookkeeping. A few APIs like `Layer.top()` are also fairly meaningless in the activation-order stacked system. Or if not meaningless, they're to be used... differently. As in, the top - visible - layer might not be the one with the highest index. That changes a lot of assumptions in the code, in various plugins.

But lets step back a little, and explore how these two work, before diving further into possibilities.

# Index-order

With index order, we simply stack layers in index order. A higher numbered layer will shadow a lower one. This sounds simple, but in reality, it can be fairly confusing. I remember that when I first encountered layers, I thought transparent keys are looked up in the nearest lower layer, active or not. I did not recognise that only active layers are considered. It's also surprising that you can't switch to a lower layer from a higher one, only by turning the higher one off. Which makes sense if you understand index-order stacking, but from a new-comer point of view, switching to another layer does not mean index-order stacking. From a cognitive load perspective, with index-order, we  have to remember the indexes of the layers we activated, which is a bit more than if we only had to remember the order of activation. At least as far as I'm concerned, I find it easier to remember I activated "Symbols" first and then "Mouse", than it is to remember that "Symbols is layer 1" and "Mouse is layer 2".

Bookkeeping in this case is simple: flip bits in an `uint32_t` layer state variable, done. Small, efficient, fast. Walking the stack is also a linear process. Letting other plugins know which layers are active are also fairly simple, and they can easily, linearly walk the stack backwards.

# Activation-order

Activation order sounds conceptually simpler on paper: whatever you activated last, is on top. But we still have to remember the order of activation, in case we have multiple layers active. We don't have to remember the indexes of layers, just the order of activation.

Bookeeping in this case, however, is a bit harder, we can't just flip bits, we need to keep a stack of `uint8_t`s (technically, we could fit two indexes into a single `uint8_t`, since we have a limit of 32 layers, but the ram we'd save that way would be greatly eclipsed by the complexity of addressing and unpacking). It's also a bit trickier when deactivating a layer from the middle of the stack (we either have to remove and rearrange the stack, or place a tombstone and GC later). We'd also need to provide slightly different APIs to plugins.

Nevertheless, activation-order is a viable strategy, and for some people and some use-cases, might even be a better choice than index-order.

# Transparency

Both cases can support transparency, but I believe it makes more sense to have transparency enabled for index-order by default, and disabled for activation-order. We should be able to control it independently, though.

# Proposal

So this is where the big idea comes in: what if we made a Layer base class, and would allow the hardware to choose a default, and the sketch to override it? For this, we'd have a `SketchProperties` struct, defined by the device with a default, and set up as a weak symbol. The sketch should then be able to override it. To make partial overrides easier, we could have something like this:

```c++
struct Model01SketchProperties {
  static kaleidoscope::driver::layer::IndexOrder Layer;
  // ...
};

struct SketchProperties __attribute__((weak)): public Model01SketchProperties {};
```

And then in the user sketch, one could do:

```c++
struct SketchProperties: public Model01SketchProperties {
  static kaleidoscope::driver::layer::ActivationOrder Layer;
}
```

Mind you, this is wholly untested, may or may not work in practice. This would allow us to have different layer implementations, all with their pros and cons.

Using these `SketchProperties` could also be useful for other things, like making it possible to disable `BootKeyboard` (when you never use it, and want to save space) or `NKROKeyboard` (when your main OS doesn't support NKRO anyway, and you want to save space) support or the like.

algernon, 2020-01-03:
Well, the sketchprops thing will need more thought, we can't just override the type of it too.

algernon, 2020-01-03:
As an intermediate step, we could provide Layer drivers, and allow devices to choose theirs. That'd already be a step up from the status quo, and once we figure out how to allow sketch-level overrides, we can wire those in too.

obra, 2020-01-03:
Layer drivers: I don't think I want to make the layers system -that-
configurable. I don't think the additional functionality is going to
to offset the complexity and user confusion.

Sketch props: I'm pretty leery of exposing the Props system to end-users.

Wouldn't going to non-transparent layers mean we could do away with the
entire LiveCompositeKeymap system?

I'm not sure it's -worth- the hassle of the change, but it seems to me like
it might end up with a net savings in terms of ram and flash.
ᐧ

algernon, 2020-01-05:
- Sketch props: Fair enough.
- Transparency: I really, really, really want to keep it, because when it is useful, it is incredibly powerful. It's already easy to not use it: don't place transparent keys on the keymap.

The bigger difference between our current system and the additional one in my proposal is stacking order. We currently stack in index-order, and my proposal had an additional variant, which stacks in activation order. The two are very different, which is why I'd like to make the layer system at least partially pluggable.

I mean, I could make the current layer code support both, with a run-time toggle even. But that would add the cost of activation-order stacking for _everyone_. With a switchable system, we only incur that cost if using activation-order.

obra, 2020-01-07:
I really don't think we want both kinds of layer activation systems. 

Hrm. I've got an idea half formed in my head about implementing MoveToLayer as locking to an all-XXX layer and then locking to the new layer. (Not asking for any action. Just writing down what's in my head)

algernon, 2020-01-07:
The problem with that idea is that it still doesn't allow stuff like `MoveToLayer(3)` -> `MoveToLayer(2)` and end up with layer#2 active, something which activation-order would support. The thing is, that index-order stacking is confusing until you get used to it, even without transparency. For disabling transparency, we can already override `Layer.getKey`, so that alone is no problem.

Hm. Hmmm. To be honest, I can probably do some weird stuff outside of Kaleidoscope by abusing `Layer.getKey`... I'll see about that. If I can, then `MoveToLayer(N)` can remain what it currently is in #779, a way to disable all other layers, with no other changes needed within the layer system. If I can implement the desired behaviour outside of core, perhaps as a plugin-like thing, that works for me.

commit 0f1db922 (2020-01-01): layers: Add a MoveToLayer(n) key

While `ShiftToLayer` and `LockLayer` activate the given layer, and keep all
others active as well, `MoveToLayer` activates the given layer, and deactivates
all others. This allows one to have discrete layers that stand on their own,
without any other layers interfering.

Basically, this is a different way to work with layers, a less powerful, but
also simpler one.

Fixes #564.
commit 5c49b2ca (2020-01-02): layers: Add a MoveToLayer(n) key

While `ShiftToLayer` and `LockLayer` activate the given layer, and keep all
others active as well, `MoveToLayer` activates the given layer, and deactivates
all others. This allows one to have discrete layers that stand on their own,
without any other layers interfering.

Basically, this is a different way to work with layers, a less powerful, but
also simpler one.

Fixes #564.
commit 55f7b6b9 (2020-01-09): layers: Add a MoveToLayer(n) key

While `ShiftToLayer` and `LockLayer` activate the given layer, and keep all
others active as well, `MoveToLayer` activates the given layer, and deactivates
all others. This allows one to have discrete layers that stand on their own,
without any other layers interfering.

Basically, this is a different way to work with layers, a less powerful, but
also simpler one.

Fixes #564.
commit 2ed92aaf (2020-01-22): layers: Add a MoveToLayer(n) key

While `ShiftToLayer` and `LockLayer` activate the given layer, and keep all
others active as well, `MoveToLayer` activates the given layer, and deactivates
all others. This allows one to have discrete layers that stand on their own,
without any other layers interfering.

Basically, this is a different way to work with layers, a less powerful, but
also simpler one.

Fixes #564.
closed 2020-02-11 by obra