keyboardio/Kaleidoscope #568: The mid-cycle report problem

gedankenexperimenter, 2019-02-13:
### Summary

It is fairly common for plugins to send keyboard HID reports in the middle of a scan cycle, typically in response to detecting a key press or release event. The problem with this is that simply calling `hid::sendKeyboardReport()` when only part of the keyswitch scan has been processed is that there might be another key that is being held, but which has not yet been processed during the current scan cycle. This means that it's not safe to base these mid-cycle reports on the (incomplete) current keyboard HID report. This is the same reason that, when checking if a modifier key is being held, we check the _previous_ report, not the current one.

If a mid-cycle report is based on the current HID report, it can easily result in sending a report that doesn't have the keycode for a key that is being held, resulting in a key release, then (on the next report) a key press event in the host OS. In the worst case, it can result in a plugin that sends a mid-cycle report in every cycle rapidly causing a keycode to toggle on and off, sending one keypress per cycle to the OS.

### An Example

This example is trivial, but I think it's a clear demonstration of the problem. Suppose a plugin (e.g. Macros) is designed to send one or more keyboard reports when a certain key (<kbd>1</kbd>) is pressed, and that key is the first one scanned. It first sends a press of `Key_A`, then a release of the same key. By itself, this works just fine, but if another key (<kbd>2</kbd>, scanned second) is still being held, perhaps just rollover, and is mapped to `Key_B` (no special plugin action), here's what happens:

#### sequence of events

1. press <kbd>2</kbd>
1. press <kbd>1</kbd>
1. release <kbd>2</kbd>
1. release <kbd>1</kbd>

#### what happens

1. `b` is added to the report. At the end of the scan cycle, that report is sent to the host, then the report is cleared. No problem.
1. `a` is added to the (empty) report (<kbd>2</kbd> hasn't been processed yet), and a mid-cycle report is sent. This report contains `a`, but not `b`, so the OS detects a simultaneous press of `a` and release of `b`. Next, `a` is removed from the report, and another (empty) report is sent. Then `b` is added to the report. At the end of the scan cycle, a report is sent to the host, which perceives a second press of `b`, even though the user didn't release it and press it again.
1. nothing is added to the report, because the plugin responsible for `a` doesn't act on held keys. The OS receives a second release of `b`.
1. both keys are released, so we're done.

The output will be `bab`, but should be just `ba`.

### An Existing Workaround

Qukeys already deals with this problem by directly accessing the `Keyboard.lastKeyReport` and `Keyboard.keyReport` member variables, which were made public for the purpose. It backs up `Keyboard.keyReport`, and replaces it with the contents of `Keyboard.lastKeyReport`. That way, any keys that haven't yet been scanned already have their keycodes added to the report, and it's safe(r) to send a mid-cycle key report. Then it restores `Keyboard.keyReport` from its backup so that the current cycle can continue as normal.

This works, but it's fragile. Qukeys bypasses the HIDAdaptor and reads and writes internal members of a KeyboardioHID class itself. It won't work with any HID module other than KeyboardioHID. And it only works with the NKRO `Keyboard` class, not the `BootKeyboard` class.

### A General Solution

If we added two functions in the HIDAdaptor, to let plugins prepare the HID module for a mid-cycle report and to restore state after, it would be much simpler for plugins to get this right, and those member variables could be kept private. I'm calling the two functions `hid::stashKeyboardReport()` and `hid::restoreKeyboardReport()`, but they would mostly just call similar methods of the HID module's `Keyboard` and `BootKeyboard` classes, as appropriate. Plugins would not need to know what HID module or type of report is being used, because the HIDAdaptor would insulate them from that.

A downside to doing it this way is that the HID `*Keyboard` classes would now need to store three HID report structures. In the case of the NKRO class, that 87 bytes of RAM (29 more than we currently use), and that needs to be allocated permanently, whereas what Qukeys currently does is storing those extra bytes only as long as necessary.

---

### An Alternative

There is, in my opinion, and even better way to solve this problem, but it's a much more radical change. This would be to build the report only when it's time to send it, from a different data structure that stores the state of the keyboard. We (essentially) already have this structure, in `Layer.live_composite_keymap_[]`, though it would need to be changed to clear entries when keys are released. It would then become possible to store only `lastKeyReport` (for comparisons), and it would obviate the need for the stash & restore functions, because the keyboard state wouldn't need to be cleared entirely every cycle. It would also be easier to get up-to-date information about the state of the whole keyboard.  There would be more ways for keys to get "stuck" on, but some plugins would be able to use that as a feature (e.g. OneShot).

I don't expect this alternative solution to be practical in the short term, and I know I've suggested it before, so I won't harp on it. I do have a working version of it in Kaleidoglyph, though, and I found that it vastly simplifies some plugins.

gedankenexperimenter, 2019-02-13:
I'll submit some PRs to demonstrate this proposal soon. There has already been a bit of discussion on this topic in #560.

obra, 2019-02-13:
Thanks so much for the detailed writeup. The description of the issue
suggests to me that the plugins in question are trying to do their work too
early.

Now that we have during-scan and post-scan hooks for plugins, would it be
possible to have the plugins that run into this issue only send their
updates at the end of the cycle?
ᐧ

gedankenexperimenter, 2019-02-13:
That's a good question. I think it's _possible_, though perhaps not in every case. But to do so would be complex and error-prone, I believe.

Plugins are generally written to respond to keypress and release _events_, and they have ready access to those events in the middle of the scan cycle, when `handleKeyswitchEvent()` is calling their event handlers. If a plugin needs to send a keyboard HID report, or more obviously, a sequence of reports, in response to a keyswitch event, it's natural and simple to take that action in its `onKeyswitchEvent()` method. It could store the event and wait until later to send those reports, but then it needs to do a bunch of bookkeeping that is unnecessary if the reports are sent immediately upon processing the event. This demands more memory for each plugin, and introduces more complexity and opportunity for errors that are harder to debug.

It also would create cases that don't need to be handled if those reports are sent mid-cycle. For example, what if a plugin needs to send extra reports originating from two different keys? Or three? As a plugin author, I now have to decide how much memory I'm willing to reserve for bookkeeping in order to handle those cases, and even if I set the limit at just one per cycle, whatever memory I reserve isn't on the stack, so it's unavailable even when my plugin is doing nothing.

I hadn't considered the idea before, so my first reaction may not be very accurate. I think it's certainly worth discussing further.

Now that I think about it, I'm pretty sure that when writing Qukeys, I did consider saving the extra reports until the end of each cycle, but ultimately rejected the idea. It may be that I had to use a fixed-size event queue, and didn't want anything crazy to happen when that limit was exceeded, necessitating queue flushes mid-cycle in that case. I suspect that wasn't the only reason, though.

gedankenexperimenter, 2019-02-13:
I've added three interdependent PRs to demonstrate my proposed solution to this problem:

- keyboardio/KeyboardioHID#49
- keyboardio/Kaleidoscop-HIDAdaptor-KeyboardioHID#16
- #570

obra, 2019-02-13:
I think I disagree about it being natural and simple. It's certainly something that seems straightforward to an implementer when they're first implementing something, but it's ultimately something that leads to a lot of complexity. Causing a key report to be issued in the middle of scanning the keyboard means that, for example, behaviour is going to be different based on the order of the keys being scanned. 

I 100% understand why it was easiest to write plugins this way when we weren't giving plugins other options for hook points. I also want plugins to be able to do crazy things with HID reports. I don't think that an implementation that mucks about with the partially-built HID feels "right" -- If we're going to do something that still has plugins sending HID reports during the key scanning process, an implementation that might make more sense to me would be to have our HID facade storing the "in progress" report and only populating the backend implementation as we're about to send it.

Longer term, I'd dearly love to have a shared global event queue that plugins are able to use and mess with.

gedankenexperimenter, 2019-02-13:
> Causing a key report to be issued in the middle of scanning the keyboard means that, for example, behaviour is going to be different based on the order of the keys being scanned.

In the corner cases, behaviour will inevitably depend on the scan order. For a plugin that needs to serialize key events in order to function, scan order is the only tie breaker that makes sense if two of them occur in the same cycle. While this may seem like splitting hairs, the changes that I'm proposing are designed to eliminate all but that unavoidable dependence on scan order – to the same extent that delaying the extra report(s) until the scan cycle is complete would.

> I 100% understand why it was easiest to write plugins this way when we weren't giving plugins other options for hook points.

The pre- and post-report hooks were available when I first started writing Qukeys. I don't know of any third-party plugins that were written before those hooks were in place. The additional hooks that are available (and the changes to the whole system, for that matter) have no effect on the additional bookkeeping required for a plugin to delay sending a report until the scan is complete, as opposed to sending a report mid-cycle, when it already has access to the `Key` value, state, and coordinates.

> I don't think that an implementation that mucks about with the partially-built HID feels "right"…

I agree 100%, but I don't consider the functions that I'm proposing to be "mucking about" with the HID report. Quite the contrary, what I want to do is provide functions so that a plugin can _preserve_ the incomplete report, and make changes based on a complete report instead before sending, then restoring that incomplete report.

> If we're going to do something that still has plugins sending HID reports during the key scanning process, an implementation that might make more sense to me would be to have our HID facade storing the "in progress" report and only populating the backend implementation as we're about to send it.

I think this idea would just move the location of the problem from one bit of code to another, without solving anything. Suppose the in-progress report is stored in the HID façade instead of the HID module itself. How does that help? As long the keyboard state (currently represented primarily in the HID report) is cleared every cycle, and rebuilt during the scan cycle, there will be a partial report that is unsafe to use mid-cycle.

> Longer term, I'd dearly love to have a shared global event queue that plugins are able to use and mess with.

Same here. That's what I did when I wrote Kaleidoglyph, and that's what my alternative solution is all about. Keyboard state is not stored in the HID report, because it can't capture all the information (two keys adding the same keycode; modifiers that are the same in the HID report, but must be treated differently internally (i.e. mod flagged key rollover), et cetera), but in the equivalent of `live_composite_keymap_[]`, where `Key_Transparent` means "idle" and `Key_NoKey` means "masked". The key scan can then send each event, upon detection to all the plugins one time, and reports are sent after each _event_, not after each scan cycle. There's never a "partial report", because the keyboard state is always up-to-date.

I'll see if I can find the time to figure out how to rewrite Qukeys to only send reports at the end of the scan cycle, and see how much complexity it really adds. Maybe I'm overestimating it, after all. I'll also try the same exercise with a version of PrefixLayer, which is a much simpler plugin that also sends mid-cycle reports.

gedankenexperimenter, 2019-02-13:
I've just made what I think are the minimal modifications to PrefixLayer to send its extra reports at the end of the scan cycle instead of mid-cycle: gedankenexperimenter/Kaleidoscope-PrefixLayer#1

For comparison, I also added the report-stashing code to protect from the mid-cycle problem: gedankenexperimenter/Kaleidoscope-PrefixLayer#2

The end-of-cycle changes are much more extensive, and the resulting plugin is considerably more complex. It also still suffers from issues that the mid-cycle (stashed) version does not. All of those issues can be dealt with by adding even more complexity to the plugin, except that there would still be a limit to how many simultaneous keypresses it can handle. That's kind of a nitpicky corner case, but the complexity of handling two is the same as the complexity of handling 20.

gedankenexperimenter, 2019-04-04:
So, another bug (#619) came up yesterday, and it's in this same area.  And the PRs that I've submitted already to address this won't fix it. I plan to have another look at possible solutions, maybe sometime soon. I expect to propose one of two different approaches:
1. An expansion of the PRs I already submitted, probably requiring deeper changes to KeyboardioHID and/or Kaleidoscope-HIDAdaptor-KeyboardioHID.
1. A more radical change, where the keyboard state information isn't stored only (or even primarily) in the HID report.
I've been using the latter option in Kaleidoglyph, and it works very well. In the long term, I think it's clearly a better solution, but in the short term, the transition could cause lots of bugs in plugins.

The root of the problem has become clearer to me today as I struggled to find a way to fix #619 – the HID report doesn't store enough state information. It works just fine for a keyboard where each key sends one and only one unique keycode, but those are not valid assumptions for Kaleidoscope.

closed 2021-05-28 by algernon