keyboardio/Kaleidoscope #569: MouseKeys: Warp to other display

qayshp, 2019-02-13:
Absolutely loving mouse keys and warping.

I find myself reaching for the physical trackpad/mouse/trackball less and less frequently, but most often when I need to quickly dart across monitors.

One way I have been able to do this without taking my hands from my Model 01 is by warping to an edge of the display, and then using the directional mouse keys to push slightly into the next display. At that point, I can again warp happily. Perhaps I can make a macro for this, but don't really know how.

algernon, 2019-02-13:
Warping to another display is something that's up to the OS, I'm afraid, we can't do much about it on the firmware side. For example, GNOME under Linux presents all displays combined as a big virtual screen, so warping around on that does cross display boundaries. As far as I remember, Windows and OSX don't do this, and warping there is restricted to a single screen. One might be able to configure them to change this behaviour - I haven't looked yet.

obra, 2019-02-13:
I’ve actually wondered if we should try to kick one pixel to the right when we end up warping to max(x)

algernon, 2019-02-13:
Could do that, yes. We'll need to reset the warp at that point I think, but it's certainly doable. (Same for `min(x|y)` too, by the way - basically whenever we reach any edge.)

KenMSK, 2022-06-27:
I am trying to write a macro to wrap to another screen. The idea is very simple. Wrap to the edge, and move left some pixels.
```
    case MACRO_LEFTSCREEN:
      kaleidoscope::Runtime.hid()
        .absoluteMouse()
        .moveTo(0, MAX_WARP_HEIGHT / 2, 0);
      kaleidoscope::Runtime.hid().mouse().sendReport();
      kaleidoscope::Runtime.hid().mouse().move(-383, 0);
      kaleidoscope::Runtime.hid().mouse().sendReport();
      break;
```
      
This actually kind of works. The problem being, the moveTo() and move() call are async. I guess such is the nature of mouse. Sometimes one get executed earlier than the other one. I cannot get them to work 100% of the time. I think the solution should be to add a little delay / await... Not sure if that is possible / easy to do.

lobre, 2023-02-02:
> GNOME under Linux presents all displays combined as a big virtual screen, so warping around on that does cross display boundaries.

I am running Linux/Xfce, and the behaviour seems to be that even simple warping is not intuitive. As you say @algernon, the chosen position seems to be taken from a geometry that is "all displays combined as a big virtual screen".

If I have two monitors side by side of the same size, and if I use a 2x2 grid and want to warp to the North East, it goes to the right monitor in the middle of the screen.

Would be more convenient to have it warp to the current monitor. But it seems undoable if I understand correctly.

Rom1deTroyes, 2023-02-03:
> I am running Linux/Xfce, and the behaviour seems to be that even simple warping is not intuitive. As you say @algernon, the chosen position seems to be taken from a geometry that is "all displays [...]

Yes, it is the way X server default configuration works : you got the same behavior with wacom table or touchscreen.

> Would be more convenient to have it warp to the current monitor. But it seems undoable if I understand correctly.

Long ago, I was using something like :
```
xsetwacom set "WALTOP International Corp. Slim Tablet stylus" MapToOutput HEAD-0
```

To map my Waltop tablet to my main screen... maybe something to search in the Xorg config ?