keyboardio/Kaleidoscope #571: TapDance gets "stuck"

tiltowait, 2019-02-13:
As detailed in [this thread](https://community.keyboard.io/t/tapdance-getting-stuck/2517), TapDance can get stuck in certain circumstances. I have a very simple configuration: double-tap left control and get `LGUI+[`. Double-tap right control and get `LGUI+]`. However, I've found that pressing `Control+Tab` will lock the keyboard into expecting another control tap for TapDance. As a result, the next time I type `Control+Tab`, my Mac will switch applications instead of switching browser tabs (and will also frequently go back a page, to boot).

My TapDance config:

```c++
void tapDanceAction(uint8_t tap_dance_index, byte row, byte col,
                    uint8_t tap_count,
                    kaleidoscope::plugin::TapDance::ActionType tap_dance_action) {
  switch (tap_dance_index) {
  // 1. Left control
  // 2. Shift+[
  case BACK:
    return tapDanceActionKeys(tap_count, tap_dance_action, Key_LeftControl,
                              LGUI(Key_LeftBracket));
  // 1. Right control
  // 2. Shift+]
  case FORWARD:
    return tapDanceActionKeys(tap_count, tap_dance_action, Key_RightControl,
                              LGUI(Key_RightBracket));
  }
}
```

gedankenexperimenter, 2021-05-26:
I think there's a good chance this is fixed by #1024. If I can reproduce it, I'll write a testcase and submit a PR.

gedankenexperimenter, 2021-05-26:
I've got a testcase that doesn't exactly reproduce this bug, but I believe it does illustrate the current state of the problem. Using the above `tapDanceAction()` code, typing `TD(BACK)` + `Key_Tab` twice in a row results in an extra press and release of `Key_LeftControl` (not generally user-visible), followed by the correct `ctrl` + `tab` on the first press. Then on the second press of the combo, there's no `ctrl` in the output; all I see is `tab`.

As expected, this is fixed by #1024.

closed 2021-05-28 by algernon