keyboardio/Kaleidoscope #572: Escape key is broken when using OneShot and Escape-OneShot

TreTuna, 2019-02-14:
When using the current master in combination with OneShots and Escape-OneShot, the Escape key is broken and Escape-OneShot works, but it doesn't send the keycode out to the host.

After a `git bisect` it seems that the culprit was [OneShot: Switch from 16-bit bitfields to using a struct](https://github.com/keyboardio/Kaleidoscope/commit/032da484bc8b55e9df450e394fae88da5e442451).

algernon, 2019-02-14:
That means I broke [one of these](https://github.com/keyboardio/Kaleidoscope/blob/aecc29a31341cd21f5d90133b5bcf9237eb981a5/src/kaleidoscope/plugin/Escape-OneShot.cpp#L31), which is entirely possible.

My suspicion is that I broke `isActive()`, but will have to debug it further. Since I can easily reproduce the problem, this shouldn't take too long.

algernon, 2019-02-15:
Turns out that `isActive()` is fine, but `isPressed()` and `isSticky()` always return true.

PR #573 (2019-02-15): OneShot: Fix isPressed() and isSticky()
I tried to be smart and use `sizeof()` instead of hard-coding the array size, but I used it wrong, and we iterated to 32 instead of 16, overflowing the array and looking at parts of the memory we had no business looking at. This resulted in `isPressed()` and `isSticky()` always being true.

Instead of trying to be clever, use `OneShot::ONESHOT_KEY_COUNT` throughout, which is a constexpr defined at 16, the number of oneshot keys we have.

Fixes #572.
commit 51a7537d (2019-02-15): OneShot: Fix isPressed() and isSticky()

I tried to be smart and use `sizeof()` instead of hard-coding the array size,
but I used it wrong, and we iterated to 32 instead of 16, overflowing the array
and looking at parts of the memory we had no business looking at. This resulted
in `isPressed()` and `isSticky()` always being true.

Instead of trying to be clever, use `OneShot::ONESHOT_KEY_COUNT` throughout,
which is a constexpr defined at 16, the number of oneshot keys we have.

Fixes #572.
closed 2019-02-15 by obra