keyboardio/Kaleidoscope #584: Building with MSYS on Windows appears to be broken

noseglasses, 2019-02-21:
Building the stock firmware on Windows with MSYS used to work a while (ten months) ago and now appears to be broken. The last successful build attempt was with Kaleidoscope 39601fcba2d7b9bdf0a98aafd26aa88505ef9330 as part of Arduino-Boards.git (unfortunately I do not have the SHA of the bundle).

I am running a MSYS-driven build with appveyor to ensure that the firmware binary that is build with my custom build system [Leidokos-CMake](https://github.com/CapeLeidokos/Leidokos-CMake) remains binary compatible with the binary generated by kaleidoscope-builder.

A build log of an old successfull build is here: https://ci.appveyor.com/project/CapeLeidokos/leidokos-cmake/builds/15594966

A current build output can be found here: https://ci.appveyor.com/project/CapeLeidokos/leidokos-cmake

The following lines caused by `avr/libraries/Model01-Firmware/Makefile` are especially strange
```
Makefile:55: C:/projects/leidokos-cmake/hardware/keyboardio/avr/build-tools/makefiles//rules.mk: Not a directory
make[4]: *** No rule to make target 'C:/projects/leidokos-cmake/hardware/keyboardio/avr/build-tools/makefiles//rules.mk'.  Stop.
```
In line 55 there is only the include directive that includes the rules.mk file.

Any ideas?

obra, 2019-02-21:
I don't think I was aware that we ever successfully build with msys.

But I'm pretty sure your issue is due to your build system not being
updated for the repo reorganization. build-tools is up a level now

Makefile:55:
C:/projects/leidokos-cmake/hardware/keyboardio/avr/build-tools/makefiles//
rules.mk: Not a directory
119 <https://ci.appveyor.com/project/CapeLeidokos/leidokos-cmake#L119>make[4]:
*** No rule to make target
'C:/projects/leidokos-cmake/hardware/keyboardio/avr/build-tools/makefiles//
rules.mk'. Stop.
120 <https://ci.appveyor.com/project/CapeLeidokos/leidokos-cmake#L120>

ᐧ

noseglasses, 2019-02-21:
That's funny. You had a build system that accidentaly enabled command line builds on another platform :-)

I had already fixed the up-one-level thing. It works fine with travis on MacOS and Linux.

obra, 2019-02-21:
Do you mean that even with that fix, windows still doesn't build? Do you
have an updated build failure log?
ᐧ

noseglasses, 2019-02-22:
Yes I already had applied the up-one-level fix.

A build failure log (manual build on Windows 10 with [MSYS2](https://www.msys2.org/)) of a failed attempt to build Kaleidoscope at the command line can be found [here](https://gist.github.com/noseglasses/07973db63b7357ebf66c95a72df4b7bb).

obra, 2019-02-22:
Yeah, it looks like the fallback case in the makefile is outdated. That I
can fix. At the same time, our standard Makefile needs to be made aware of
win32 in its various flavors. That...I don't know enough to fix.
ᐧ

obra, 2019-02-22:
(I pushed the change to Model01-Firmware's makefile)
ᐧ

noseglasses, 2019-02-22:
I will check if this f

noseglasses, 2019-02-23:
Builds with MSYS2 now work again. Thanks.

algernon, 2019-03-09:
Since the build works now, I'm closing the issue.

closed 2019-03-09 by algernon