keyboardio/Kaleidoscope #586: Feature Request: Hardware specific key coordinate constants

noseglasses, 2019-02-22:
Given we had a struct `KeyCoord` that would be a wrapper for `row` and `col`, we could have constants like
```cpp
static constexpr KeyCoord cKeyCoordHomeRowLeftForefinger = ...
...
static constexpr KeyCoord cKeyCoordUpperRowLeftForefinger = ...
...
static constexpr KeyCoord cKeyCoordThumbKeyOne = ...
...
```
that are defined by the Hardware class. This would make it easier to access specific keys in a hardware agnostic fashion and would e.g. be valuable for those people that use Kaleidoscope on different keyboards but want to share a common sketch on all of those. This would be even more valuable when all those functons/methods of the core firmware that operate on `row`/`col` information could also be passed a `KeyCoord`. Those `KeyCoord` based accessors could easily be inlined wrappers to the existing functions.

obra, 2019-02-22:
I'm somewhat dubious that we could end up with such a mapping that's "Sane"
for keyboards as different as the Model 01 and a 104 key keyboard and an
Atreus.

But if you wanted to try to flesh out a proposal, I'd be happy to look at
it more.

I wonder if constants for "the key that means x on a 104 key keyboard"
would be a more useful mapping.
ᐧ

noseglasses, 2019-02-22:
Yeah you are right. It's definitely not possible to have common subset for all possible keyboards, some of which could e.g. one handed. But to let the individual hardware classes define their specific set of KeyCoords would already be quite useful.

noseglasses, 2019-02-22:
> for "the key that means x on a 104 key keyboard

Anything that relieves the user from writing keycodes would be very useful. For the Model01 the thumb and palm keys would be natural candidates.

obra, 2019-02-22:
have you looked at how input Club’s kll does things like this?

alternatively, can you sketch out an example showing how this would be useful across multiple keyboards? Within a single keyboard, I don’t think it is more useful than having an agreed-upon standard for the order of keys.

I think I understand your desire, but I’m having a hard time visualizing a possible solution that doesn’t end up more complicated than not having a solution.

noseglasses, 2019-02-22:
One typical application is the specification of Qukeys. Those currently must be specified explicitly, thereby providing key coordinates. Another application where `KeyCoord` constants would be beneficial is lighting up specific keys.

obra, 2019-02-22:
to me,
[cKeyCoordUpperRowOuterLeftPinkieFinger,
cKeyCoordUpperRowInnermostLeftIndexFinger, cKeyCoordPalmKey]

feels pretty unwieldy. And hard to write, since each keyboard is going to
have to have different named constants. You're also going to have a harder
time reasoning about the names if you have different opinions on touch
typing than the original author of the names.

[R0C0, R0C7,R3C7] does require counting, but is generic (so reusable) and
pretty unambiguous.

However, if you think that you've got good ideas for naming, it's something
you ought to be able to try out by creating a plugin folks can include in
their sketches to get a feel for it. If it's something that proves popular
that way, that'd be a pretty good way to convince me that such names are
useful.

ᐧ
ᐧ

noseglasses, 2019-02-23:
I agree about the keys in the typical upper, home and lower rows, while for the thumb and palm keys of the M01 to me some constants would be useful.
Given we had a KeyCoord class, it could be left to the user to define the constants she/he needs.
BTW, how do you feel about the name `KeyCoord` for such a class? Or would `KeyPos` be a better choice?
The class name `Key` is unfortunately already used for something that I would rather call something like `KeyAction`.

gedankenexperimenter, 2019-02-23:
I've been using the name `KeyAddr` for a long time now in Kaleidoglyph, so I'll add that as a suggestion. I use it instead of separate row & column numbers.

I think I like the `KeyAction` name, by the way. It's difficult with so many things that can reasonably use the term "key" .

noseglasses, 2019-02-23:
`KeyAddr` sounds like a pretty good candidate, too.

algernon, 2019-03-09:
I'll +1 the `KeyAddr` name. Hardware-specific coordinate constants however is not something I'd add, because of some corner cases: what if we have a keyboard where the index finger is responsible for multiple columns? `InnermostLeftIndexFinger`, `MiddleLeftIndexFinger`, `OutermostLeftIndexFinger`? What if we only have two? Or four such columns?

I think consistent numbering is considerably easier to understand and reason about, and is far easier to make portable across keyboards. Not perfect, but more consistent than hardware-specific naming, in my opinion.

gedankenexperimenter, 2019-03-09:
I think it would be beneficial to have a briefer notation system that is similar to that used for beginning piano students. In that case, each note in a piece of music gets labelled with a number indicating which finger should be used (thumb = 1, pinky = 5).

On a keyboard, one could start with a letter (either `R` or `L`) to indicate which hand is generally expected to type a given key. Next, a row label (either a number or letter) representing the row. I'd suggest a letter, with `H` indicating the home row, and the other rows alphabetically relative, so the row below the home row would be `G`, and above it would be `I`. Then a column label, perhaps a number this time, relative to each finger's home key, and using the same finger numbers piano students use, so `2` would be the home column of the index finger, and `5` for the pinky. That would be extensible enough even for the Model01, with its extra inner column being `0`. If there are even more columns than that, or keys are arranged in a way that simply isn't clear, it would probably need to be done in a hardware-specific way, regardless.

As for the thumb keys on the Model01, I'd probably give them a number instead of a letter for the row to indicate that they don't neatly fit into the same grid as the finger keys.

algernon, 2019-03-09:
Now, _that_ notation seems useful, and it does have advantages over the current row+column numbering, because it's all relative to the home row and finger positions.

How about this scheme: `[Hand][Row][Finger][<finger offset>]`? Where `<finger offset>` is optional, used for fingers that address multiple columns. It would always be `>= 0`, even for the left pinkies, the direction is context-sensitive.

For example, lets imagine a keyboard half with 7 columns, not unlike the Model01: pinkies address two columns, indexes address three, so on the left half, the coordinates would be `LH51`, `LH50`, `LH4`, `LH3`, `LH20`, `LH21`, `LH22`.

The problem part is the thumb cluster and the palm key. On the Model01, we can pretend that the home position of the thumb is the innermost key on the arc, and then numbering is fine. But what about the palm key? `LHPalm`? or `LH6`?

Considering an ErgoDox... I think the home position of the thumb is not the innermost key there, perhaps we need a direction there afterall: `LH1L1` for "left homerow for thumb, one to the left"?

gedankenexperimenter, 2019-03-09:
The "home" position of the thumb on the Model01 should certainly be the key with the homing dot on it. But then your "offset" numbering doesn't work, since it doesn't allow negative numbers. An alternative would be to treat the thumb clusters as columns: `LI1`, `LH1`, `LG1`, `LF1`. I would treat them as something else entirely, myself, since they're very hardware-specific, and regardless of what you do, it doesn't extend to any other keyboard.

noseglasses, 2019-03-10:
@gedankenexperimenter and @algernon, i like your ideas regarding the naming scheme for keys around home row. This sounds like something that could be used for the majority of all keyboards. This will be especially useful for those people that want to share a sketch between different keyboards.

I am not sure though, if the same makes sense for thumb/palm keys. For those, number and position are far from standard across devices. My above suggestions for naming thumb key constants was more about (non-standard) names that individual keyboard hardware headers could supply. Many (most?) people concentrate on only one keyboard, anyway. For them I guess it would be ok to look up the specific thumb/palm/etc. constants provided for their keyboard.

algernon, 2019-03-10:
Or we could just not have other constants for the thumb & palm keys than the `RxCy` ones we have now. That way, we'd have both the `RxCy` coordinates for all keys, and the homerow & finger-based ones for most keys. Since thumb & palm are special anyway, the `RxCy` ones might be a better fit there, since it's going to be different from board to board anyway.

noseglasses, 2019-03-10:
> Since thumb & palm are special anyway, the RxCy ones might be a better fit there, since it's going to be different from board to board anyway.

Agreed.

gedankenexperimenter, 2019-03-10:
> Since thumb & palm are special anyway, the `RxCy` ones might be a better fit there, since it's going to be different from board to board anyway.

I disagree. I don't see any reason to _require_ such designators for boards with keys that don't fit logically (the Function key row on a "standard" 104/5 key keyboard also qualifies) into the home-row-relative grid(s), but I see no reason to forbid it, either. And with the scheme that we're discussing, with a leading `R` meaning "right side" for most keys, but "row" for a few special ones, it seems silly to not have the Model01 hardware definition provide something for those other keys that at least matches the mirror coordinates for the finger keys. I do agree that, as it's very hardware specific, it shouldn't be part of a generic standard.

algernon, 2019-03-16:
I wouldn't forbid having custom names for those keys, just wouldn't encourage them.

With regards to naming, we'll want to have *both* the `RxCy` scheme *and* the new home-row-relative names. There are situations where the row/col coordinates make more sense (eg, LED Chase, or positioning the AlphaSquare fonts, Stalker, etc), and other situations where the relative ones are more useful. Thus, we should have both.

`RxCy` slightly conflicting with the right-hand keys does present a problem, a conflict we should resolve one way or the other.

gedankenexperimenter, 2019-03-16:
> There are situations where the row/col coordinates make more sense (eg, LED Chase, or positioning the AlphaSquare fonts, Stalker, etc), and other situations where the relative ones are more useful. Thus, we should have both.

`<nitpick>`
LEDChase uses the LED coordinates, not the key coordinates; it would be unaffected. Stalker just deals with coordinates of keys that have been pressed, so it doesn't really what notation the user uses. AlphaSquare (and Wavepool) need to know things about where the keys are physically positioned on the keyboard relative to each other, but does that depend on the notation a user would use to identify the keys? I'm not sure.
`</nitpick>`

obra, 2019-03-17:
I’d really like to see this idea get fully fleshed out and used as a plugin for a while before we think about merging it.