keyboardio/Kaleidoscope #588: Virtual builds appear to be broken

noseglasses, 2019-02-26:
While the following works on Ubuntu 18.04 to build the latest firmware for the M01
```
export ARDUINO_PATH=<place where your Arduino installation lives>
mkdir -p $HOME/kaleidoscope/hardware

git clone --recursive https://github.com/keyboardio/Kaleidoscope-Bundle-Keyboardio.git \
   $HOME/kaleidoscope/hardware/keyboardio
cd $HOME/kaleidoscope/hardware/keyboardio/avr/libraries
cd $HOME/kaleidoscope/hardware/keyboardio/avr/libraries/Model01-Firmware
BOARD_HARDWARE_PATH=$HOME/kaleidoscope/hardware make
```
replacing the build command
```
BOARD_HARDWARE_PATH=$HOME/kaleidoscope/hardware make
```
with
```
BOARD_HARDWARE_PATH=$HOME/kaleidoscope/hardware BOARD=virtual make
```
yields the following error
```
BOARD_HARDWARE_PATH="/home/noseglasses/kaleidoscope/hardware" /home/noseglasses/kaleidoscope/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder build-all
Building output/Model01-Firmware/Model01-Firmware (0.0.0-gv1.22-24-g21f9) ...
Board virtual (platform avr, package keyboardio) is unknown
/home/noseglasses/kaleidoscope/hardware/keyboardio/avr/build-tools/makefiles//rules.mk:80: recipe for target 'build-all' failed
make: *** [build-all] Error 1
```

noseglasses, 2019-02-26:
BTW, the documentation (README.md) of `Kaleidoscope-Hardware-Virtual` appears to be outdated after the monrepo merge. As far as I can see, setting symlinks like explained in the docs is no more required. Am I right?

noseglasses, 2019-02-26:
```
BOARD_HARDWARE_PATH=$HOME/kaleidoscope/hardware FQBN="keyboardio:x86:virtual" make
```
at least starts a build. But then there are compiler errors with respect to header search paths.

noseglasses, 2019-02-26:
```
BOARD_HARDWARE_PATH=$HOME/kaleidoscope/hardware FQBN="keyboardio:x86:virtual" make
```
only works when commenting the suspicious lines 215-217 in kaleidoscope-builder
https://github.com/keyboardio/Kaleidoscope/blob/7366fea5c09e7f0b58a9b6a3f047f0e20edb814d/bin/kaleidoscope-builder#L215

noseglasses, 2019-02-26:
When trying a virtual build with a minimal firmware
```
#include "Kaleidoscope.h"

KEYMAPS(

  [0] = KEYMAP_STACKED
  (___,          Key_1, Key_2, Key_3, Key_4, Key_5, Key_LEDEffectNext,
   Key_Backtick, Key_Q, Key_W, Key_E, Key_R, Key_T, Key_Tab,
   Key_PageUp,   Key_A, Key_S, Key_D, Key_F, Key_G,
   Key_PageDown, Key_Z, Key_X, Key_C, Key_V, Key_B, Key_Escape,
   Key_LeftControl, Key_Backspace, Key_LeftGui, Key_LeftShift,
   ___,

   M(MACRO_ANY),  Key_6, Key_7, Key_8,     Key_9,         Key_0,         LockLayer(NUMPAD),
   Key_Enter,     Key_Y, Key_U, Key_I,     Key_O,         Key_P,         Key_Equals,
                  Key_H, Key_J, Key_K,     Key_L,         Key_Semicolon, Key_Quote,
   Key_RightAlt,  Key_N, Key_M, Key_Comma, Key_Period,    Key_Slash,     Key_Minus,
   Key_RightShift, Key_LeftAlt, Key_Spacebar, Key_RightControl,
   ___)
) 

KALEIDOSCOPE_INIT_PLUGINS();

void setup() {
  Kaleidoscope.setup();
}

void loop() {
  Kaleidoscope.loop();
}
```
arduino-builder tries to compile `Colormap.cpp` which then tries to pull in `EEPROM.h` which is not available for virtual builds. That's crazy as there is no reference to the Colormap plugin in the firmware sketch.

Build error:
```
g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_VIRTUAL -DARDUINO_ARCH_X86 -DUSB_VID= -DUSB_PID= '-DUSB_MANUFACTURER="Kaleidoscope"' '-DUSB_PRODUCT="Kaleidoscope Virtual Keyboard"' -std=c++11 -Woverloaded-virtual -Wno-unused-parameter -Wno-unused-variable -Wno-ignored-qualifiers -DKALEIDOSCOPE_HARDWARE_H="Kaleidoscope-Hardware-Virtual.h" "-I/home/noseglasses/kaleidoscope/hardware/keyboardio/x86/cores/virtual" "-I/home/noseglasses/kaleidoscope/hardware/keyboardio/x86/variants/virtual" "-I/home/noseglasses/kaleidoscope/hardware/keyboardio/x86/libraries/Kaleidoscope/src" "-I/home/noseglasses/kaleidoscope/hardware/keyboardio/x86/libraries/Kaleidoscope-Hardware-Virtual/src" "-I/home/noseglasses/kaleidoscope/hardware/keyboardio/x86/libraries/KeyboardioHID/src" "/home/noseglasses/kaleidoscope/hardware/keyboardio/x86/libraries/Kaleidoscope/src/kaleidoscope/plugin/Colormap.cpp" -o "/tmp/tmp.zI0O18bg7w/preproc/ctags_target_for_gcc_minus_e.cpp"
/home/noseglasses/kaleidoscope/hardware/keyboardio/x86/libraries/Kaleidoscope/src/kaleidoscope/plugin/Colormap.cpp:21:10: fatal error: EEPROM.h: Datei oder Verzeichnis nicht gefunden
 #include <EEPROM.h>
          ^~~~~~~~~~
compilation terminated.
Multiple libraries were found for "Kaleidoscope.h"
 Used: /home/noseglasses/kaleidoscope/hardware/keyboardio/x86/libraries/Kaleidoscope
 Not used: /home/noseglasses/kaleidoscope/hardware/keyboardio/avr/libraries/Kaleidoscope/src
Using library Kaleidoscope at version 0.0.1 in folder: /home/noseglasses/kaleidoscope/hardware/keyboardio/x86/libraries/Kaleidoscope 
Using library Kaleidoscope-Hardware-Virtual at version 0.0.0 in folder: /home/noseglasses/kaleidoscope/hardware/keyboardio/x86/libraries/Kaleidoscope-Hardware-Virtual 
Using library KeyboardioHID at version 0.0.1 in folder: /home/noseglasses/kaleidoscope/hardware/keyboardio/x86/libraries/KeyboardioHID 
exit status 1
/home/noseglasses/kaleidoscope/hardware/keyboardio/build-tools/makefiles//rules.mk:80: recipe for target 'build-all' failed
make: *** [build-all] Error 1
```

obra, 2019-02-27:
The x86 virtual implementation is something that has never gotten the
attention it deserves. I'd really love to see that change.

I spent some time on it when I was integrating it into
Kaleidoscope-Bundle-Keyboardio, but it was mostly stabbing in the dark.

I agree that hardcoding :avr: in the board type looks...wrong, but at the
same time, I think you want to be setting the FQBN and not the BOARD, since
BOARD may not be enough to uniquely identify a board.

For now, I'd love it if we could leave the fact that arduino-builder is
"non-optimal" aside. No, it shouldn't be compiling the Colormap plugin, but
the fact that it fails due to missing EEPROM exposes an issue with the
virtual core. We -are- going to need at least a stub implementation of the
bits of the Arduino API we use in the X86 implementation. That'll include
EEPROM.

I know you got a fairly functional testing setup set up with the x86 core
quite a long time ago. I'd love to get at least a minimal test harness
checked into the core, so that we can experiment with the best ways to
write and run tests, leading to this never breaking again.

I -think- that the minimal thing I want for the x86 core is to compile to
an executable that listens to STDIN for physical key reports. (Say, a
string of 1s and 0s, with 0 representing "not pressed" and 1 representing
"pressed")

For v0, the output would be USB HID reports (to STDOUT), debugging messages
(if any) to STDERR (though those can come later). We'd eventually want a
way to output LED state too.

The executable would listen to STDIN until it sees a newline. Then it would
run a single 'loop' cycle. Then it would output as described above to
STDOUT and STDERR.

We can use that to build tests. And then we can automate the tests. And
then life gets a lot better.

ᐧ

noseglasses, 2019-02-27:
> I -think- that the minimal thing I want for the x86 core is to compile to an executable that listens to STDIN for physical key reports. (Say, a string of 1s and 0s, with 0 representing "not pressed" [...]

AFAIK, most of this was already part of Craig's original design. Once the virtual hardware will compile again there's probably not too much left to do to enable what you imagined and described above.

The recent days I have been working on https://github.com/keyboardio/Kaleidoscope/pull/589 which comes with quite an amount of changes to the Kaleidoscope core. A proper test bench that automatically  tests all parts of the firmware would make such delicate changes much more comfortable.

noseglasses, 2019-02-27:
I used the smoke-examples make target. It builds for all sorts of hardware. That's already very helpful. It would be even better, if the smoke tests would come with an individual makefile. That would enable calling `make help` to get a list of targets and e.g. `make smoke-atreus`, to build for a specific hardware. The problem with the current design is that if if one of the later tests fails, all the tests that run before have to be repeated over and over during fix-test-cycles.

Is this worth a separate issue report or will the answer be anyway "The new CLI build system will fix this"?

obra, 2019-03-06:
A new issue for that would not be amiss. The perf improvements hopefully
make it hurt a little less.

I don't think I want makefile targets per device, because that will add
some maintenance overhead. but possibly some way to use an environment
variable or a make -f invocation (as we've talked about elsewhere)

obra, 2019-03-06:
Looking at this issue in more detail again, I'm not sure the thing you're
describing -should- work.

Simply setting BOARD to virtual is not enough to tell kaleidoscope and
arduino that they should be using the x86 virtual core.

At the very least, you'd need to be setting BOARD to virtual and FQBN to
keyboard:x86:virtual

Additionally, the virtual core no longer has all the features the
Model01-Firmware depends on.

Additionally, it's not entirely clear to me if the 'virtual' core is really
intended to be a virtual model 01 with all its specific layout quirks and
devices or a generic virtual kaleidoscope keyboard. If it's the former,
then it ought to get renamed to say that. But I think we'd be better off if
we had the latter, a generic virtual keyboard core that needed its own
sketches rather than trying to reuse other device-specific sketches.

I would love to see this remedied. To make it go, I believe we need:

* An 'Arduino' x86 core
* An "instrumented keyboard" device implemented on  top of the x86 core
* Both things fully moved into the Kaleidoscope repo
* smoke tests that build the instrumented core
* the start of functionality tests using that core

noseglasses, 2019-03-06:
I just tried to build it the way it used to work before the mono-repo-fusion, as explained in the [README.md](https://github.com/keyboardio/Kaleidoscope-Hardware-Virtual/tree/8eb334da21af1c24b5838edc3bb229af3485e97a) of Kaleidoscope-Hardware-Virtual. Before, setting only BOARD to virtual worked.

> At the very least, you'd need to be setting BOARD to virtual and FQBN to keyboard:x86:virtual

i tried that (see my above posts) but it did not work.

> Additionally, it's not entirely clear to me if the 'virtual' core is really intended to be a virtual model 01 with all its specific layout quirks and devices or a generic virtual kaleidoscope [...]

AFAIK, craigdissel's virtual hardware was designed to be a virtual Model01. Back when Kaleidoscope was pretty much targeted to the Model01, that made sense.

Having a generalized virtual hardware that can simulate any keyboard would be great. As far as I can see, that would require an additional level of abstraction between the virtual hardware and the physical hardware that it is supposed to simulate.

The following is a rough sketch of what I would start with:

The virtual hardware header would be informed about the physical hardware's header file through a pre-defined macro `KALEIDOSCOPE_PHYSICAL_HARDWARE_H`. If would include it, query the physical hardware class' static members (`HARDWARE_IMPLEMENTATION::matrix_rows`, ...)
 to configure the virtual hardware class and then redefine `HARDWARE_IMPLEMENTATION` to point to the virtual hardware class. The proper keymap setup macros then are automatically available.

A CLI build would be triggered as
```
VIRTUAL_BOARD=model01 make
```
Kaleidoscope-builder would learn to check if a `VIRTUAL_BOARD` is defined and set `BOARD=$VIRTUAL_BOARD` and `FQBN=keyboardio:x86:$VIRTUAL_BOARD` accordingly. A virtual version of every supported keyboard that would be listed in `x86/boards.txt` would pass the appropriate  `KALEIDOSCOPE_HARDWARE_H` (here the virtual hardware's header) and `KALEIDOSCOPE_PHYSICAL_HARDWARE_H` to the compiler.

What then remains to be done is to apply any necessary changes to the existing virtual core and possible modifications of the I/O behavior.

noseglasses, 2019-03-06:
To enable wrapping the virtual hardware simulator binary, like I do in my setup, it would be cool to also enable to build a .so-version that comes without a main() and the I/O driver stuff.

obra, 2019-03-07:
Being able to simulate any keyboard is certainly a desirable goal, but may
be a good "2.0" goal -- It's significantly more complicated than having a
generic 104 key keyboard implementation that targets the simulator, which I
think is going to be a lot easier to work with for unit tests.

It'd be really nice to decompose the problem into "having an x86 core and
keyboard we can run tests against" and "being able to simulate any keyboard
device"

noseglasses, 2019-12-06:
Virtual builds have recently been restored and generalized to simulate all keyboards. Closing this.

closed 2019-12-06 by noseglasses