keyboardio/Kaleidoscope #590: Speeding up kaleidoscope builds CCACHE

noseglasses, 2019-02-27:
With the current build system the smoke-examples take a long time to finish. This could possibly be sped up using [ccache](https://ccache.samba.org/). ccache is a wrapper compiler that maintains a cache of compiled objects. When a new object is supposed to be compiled, ccache checks the cache to find if the same file with all included headers and command line flags has been compiled before. If so, it just takes a copy of the cached object instead of doing an actual compile. Any newly compiled objects are added to the cache. 

To use ccache, it must be passed instead of avr-gcc. The build system would additionally need to define
```
export CCACHE_PREFIX=<path_to_avr-gcc>
```
that's it.

Appart for the long running smoke-example tests this would help to speedup ordinary development builds, too.

This enhancement would need to be made optional as ccache on most platforms needs to be installed as a separate package.

obra, 2019-02-27:
Yes, this would be lovely.
ᐧ

noseglasses, 2019-02-28:
Just tried ccache with Arduino or the Kaleidoscope build system, respectively. Integration is possible in a fairly non-intrusive way (from the perspective of Kaleidoscope).

It works by replacing avr-gcc and avr-g++ with wrapper scripts that prefix the actual compiler with ccache. Thus nothing needs to be changed in any parts of the build system and the same works with any other Arduino project.

[Here](https://gist.github.com/noseglasses/f627498a62472d4c592d3760baa6d439) is a link to a gist of a bash script that automates patching Arduino to work with ccache. Before you run this script (on linux or a suitable environment) make sure that ccache is installed on your system.

Compile times on my old laptop for the stock firmware of vanilla Kaleidoscope-Bundle-Keyboardio are

|Build | Duration |
|--|--|
| First | 2:19.51 |
| Follow ups | 0:21.73 |

In my setup compile times go down by a factor of seven. But this is only using one core of the machine. AFAIK newer versions of Arduino (>1.8.5) do support parallel builds through the `-jobs` flag.

It wouldn't require much to patch kaleidoscope-builder to check the installed arduino-version for its version and then enable parallel builds if `-jobs` is supported.

noseglasses, 2019-02-28:
Travis does support using ccache as well. Using the same technique as described above could help to speed up integration testing as well and save the climate as well.

obra, 2019-02-28:
Yes. I think everyone is 100% onboard with getting ccache running. In the
past, I'd installed it locally rather than adding it to the buildsystem.
ᐧ

noseglasses, 2019-02-28:
Wow, I just tested Arduino 1.8.8. By default it uses all available cores to build the firmware and comes with gcc 5.4.0 instead of 4.9.2 (Arduino 1.8.5). Now I get completely different build times.

| Build | Duration |
|---|---|
| Without ccache |  ~0:13 |
| With ccache first/rebuild | ~0:16 |

Now ccache even slows down things a bit. That's more than surprising. But it's pretty cool that Arduino finally learned to build with decent speed.

Can anybody qualitatively reproduce these results?

obra, 2019-02-28:
This is just a plain "make" at the top level of the Kaleidoscope checkout?
ᐧ

noseglasses, 2019-02-28:
I clone the bundle an the build from the Model-01 firmware repo with
```
BOARD_HARDWARE_PATH=$PWD/../../../.. make
```

obra, 2019-02-28:
I've been poking at this all day. It looks like ccache is failing on cache
hits if we don't have a stable build directory name because .d files end up
with the name of the build directory in them.

The Model01-Firmware sketch keeps one around, but that's not 'standard'  -
I'm trying to work through whether this is something we can fix with ccache
configuration or not.

At the same time, I'm trying to get the arduino "build cache" that caches
the platform core to work for me, but haven't yet seen it actually use a
cached core build when building from the commandline.

ᐧ

obra, 2019-03-01:
The reason that the Arduino core caching doesn't work for us is that it
checks to see if there are any files in the core that have changed. This
includes things in the examples directory And we drop build output into the
examples directory. So that's going to have to change to make this not hurt
ᐧ

obra, 2019-03-01:
I've got a PR incoming with my take on ccache, along with some other
changes that allow us to better take advantage of Arduino's code caching.
ᐧ

obra, 2019-03-01:
@noseglasses Have a look at Kaleidoscope#591 ?

algernon, 2019-03-09:
#591 addresses this, I believe, closing as such. Do reopen if I'm mistaken.

closed 2019-03-09 by algernon